MyRobotLab/myrobotlab #8: Create unit testing framework.

kwatters, 2015-01-11:
The goal here is to add junit to the build so it can provide some basic unit testing for the MRL project.  They should be added in as a target to the current ant/ivy build script.

Ultimately, we should create our own subclass for unit tests that will make it easy for others to write their own tests.

supertick, 2015-01-11:
assigned to captain kwatters

kwatters, 2015-01-12:
created a "test" folder for junit tests. moved some stuff into that folder. updated build.xml so it has a "test" target   "ant test" will run the unit tests.  When deploying to travis, there seems to be a problem running the unit tests...  The tests pass locally, so I think it's environmental with the travis build.  

I need some helper elfs to let me know if there's a way to inspect the artifacts of the build if the build fails..

kwatters, 2015-01-26:
I think the basic ant stuff is in place now for this, as junit tests are running as part of the build.  I'm going to close this issue.

closed 2015-01-26 by kwatters