MyRobotLab/myrobotlab #12: Control System Sound Volume in MRL

supertick, 2015-01-12:
Suggested by Cwright:
InMoov might benefit from changing volume based on "emotional" state.
http://stackoverflow.com/questions/953598/audio-volume-control-increase-or-decrease-in-java
There might be some complexity regarding the Speech service since it primarily plays mp3 files and I don't think the stackoverflow's example is relevant.  But I will investigate.

kmcgerald, 2015-01-12:
That way when you ask him why he murdered Dr. Lanning he can pump up the volume to yell "I DID NOT MURDER HIM!"  :)

hairygael, 2015-01-12:
Yeah that could be handy in some cases

LunDev, 2015-01-12:
Volume! Wanted to ask for that for a long time, always forgot it.
In this context, is there already a function to pause & resume an audio file? @supertick

supertick, 2015-01-13:
Some work would be needed to support pause - The Java Zooms AdvancedPlayer is in the AudioFile Service - it would need exposing - like this - http://stackoverflow.com/questions/16882354/how-to-play-pause-a-mp3-file-using-the-javazoom-jlayer-library

moz4r, 2017-05-02:
I saw poly use some sound volume tweak but it is not generic thing. @MatsMRL added support too to an i2c adafruit hardware amplifier

supertick, 2017-08-28:
Volume & Pause would both be good additions (should be separate enhancement requests)

moz4r, 2018-07-13:
ohh we have generic mouth / audiofile setVolume now. enough to close ?