MyRobotLab/myrobotlab #14: Python service doesn't handle CRLF nicely

kwatters, 2015-01-26:
If a python script has windows style line endings , the python service fails to parse it properly with the following error message.

51637 [Thread-9] ERROR class org.myrobotlab.logging.Logging  - ------
SyntaxError: ("mismatched character '\\r' expecting '\\n'", ('<string>', 2, 53, '\n'))

kwatters, 2015-01-27:
I added something that will make this work "better" , however, if you use the python syntax of
x = """ some arbitrary string here """    it might scrub some of the line endings in that quoted python text...  not sure what other edge cases I may have created by auto-translating \r\n  to just \n

supertick, 2015-01-27:
Nice !  I still would like to find root cause, I suspect the GUI's saving or importing is somehow foobar'ing the script up.

kwatters, 2015-01-27:
Yup, I had seen this issue many times when I first started trying to use MRL.  It seemed to happen when trying to cut & paste stuff into the MRL python editor, then when MRL would exit, I believe it saves the scripts.  I think maybe that was the reason for the added CRLF's.   I tried that this time, but I didn't reproduce it.  So, I crafted a small python file and manually added the ^M 's to the last lines to reproduce the issue.  It seems to be gone now.  (seems like something worth of a unit test :) ) 

So, the save on exit could be doing this, but this fix will handle the case where the original file has the ^M's at the end of each line.  If I've got some more time to dig into it, I'll look into the gui & the save functionality to see if that adds the extra \r's

supertick, 2015-05-03:
is this closed ?

kwatters, 2015-05-03:
yeah,  I think we can close this.  I haven't seen this in a while.  I don't know if we made any specific changes regarding this, but it seemed to be working properly in 1.0.115 with the wevolver folk.

closed 2015-05-03 by kwatters