MyRobotLab/myrobotlab #16: Runtime Dependency on Jython

jeffrschneider, 2015-02-09:
Some services (like InMoov) have a dependency on having Jython installed, however, the base install of mrl doesn't require it. New users see a PyObject error on fail. Either make it a precondition for a successful mrl installation, or have it loaded after the fact.

supertick, 2015-02-09:
This is/was the tip of the IceBerg on a huge refactoring change in dependency management.
The small change was changing dependency description data format from xml to json.
The BIG change was to normalized transitive dependencies !

Previously - the service (and other composite services) like InMoov would need to list out all of their dependencies explicitly in https://github.com/MyRobotLab/repo/blob/master/serviceData.json 

Now - the Peer DNA (Description of Neighboring Automata - http://myrobotlab.org/content/sample-inmoov-head-dna) data of each composite Service is linked to the transitive dependency management system.  

Previously the InMoov service had to have every third party library listed in the serviceData.json, now it has none listed, because it does not directly depend on them.  Rather, it depends on sub/peer services which eventually have third party library dependencies - this normalizes the references. There is less maintenance and fewer incorrect references.

I have fixed these thing locally and tested - but will keep the ticket open until the next release.

supertick, 2015-04-26:
Fixed & Tested in 114

closed 2015-04-26 by supertick