MyRobotLab/myrobotlab #17: GUI on Mac loses the mouse position

jeffrschneider, 2015-02-09:
The GUI isn't working on a Mac. Specifically, the issue was seen on Mac Air Yosemite release. 

The issue is the mouse pointer loses its position soon after launch. Attempts to click on things fail, no right clicks to launch, etc. Currently, the issue renders the GUI unusable on a Mac. 

Suggestion is to (temporarily?) take it offline. If a user attempts to run on Mac, check OS, notify user it isn't supported and terminate. Longer term, you can fix this - or just do Web UI.

kwatters, 2015-02-09:
I see the GUI on windows lock up sometimes when starting a bunch of services.  I find that clicking the maximize button and then un-maximize button refreshes the screen and it starts responding to clicks again.  This was on windows.  Maybe the same behavior happens on Mac?

Kakadu31, 2017-05-03:
[MRL_report.pdf](https://github.com/MyRobotLab/myrobotlab/files/974442/MRL_report.pdf)
I tested around with the GUI and I found some errors which i documented in this PDF.
Also send no worky ( all written in the PDF)
I hope someone knows more about the GUI than me.

supertick, 2017-08-27:
I have put in several fixes into the SwingGui .. one of the biggest culprits was the SwingGui service itself, which re-packed after a new service was started.  I removed this and I have virtually no more "locking up" on windows.  But this ticket was about macs and I don't have one to test on.  

Any Mac testing would be appreciated.

Thanks.

Kakadu31, 2017-08-27:
I can do some tests on my mac if I have time tomorrow.

Kakadu31, 2017-08-28:
I tested with version 2430 on my MacBook Pro, OS X El Capitan and couldn't produce any error, even those i found a few months ago. So I would say we can close this ticket for now!

closed 2017-08-28 by Kakadu31