MyRobotLab/myrobotlab #19: Adafruit motor controller seem to be no worky for mastab/

kwatters, 2015-02-13:
looking at the arduino code,it seems like the MRL commands for controlling the Adafruit motor controller are not defined in the MRLComm.ino.

supertick, 2015-02-21:
Yeah .. so it did work .. but I don't like the way it was done previously ..
It used Adafruit's library ... which I'm sure is a good library, but it added both a dependency to get compiled into MRLComm and additional code to be added/merged to MRLComm.ino which other systems did not need ..

I think optimally, the Adafruit library needs to be decomposed and the Motor in MRL needs to handle the logic to move the motors correctly using the simple low level commands of pinMode, digitalWrite, analogWrite etc.. of MRLComm.ino .. this eliminates all the complexity of the other dependencies.

Looking at the situation of the two Services Arduino & Motor & the 2 interfaces MotorControl and MotorController .. I've come to the belief that Motor should deal with the details - MotorController interface should be removed & there should be a "new" interface MicroController which allows the getting, reading, and digital/analog writing of pins....   and thats a good hunk of refactoring :)

MatsMRL, 2017-05-02:
This part has already been refactored quite a bit. 
All the Adafruit16CServoDriver logic has been moved from MRLComm to the Adafruit16CServodriver.
All that Arduino / MRLCom is doing is a passthru for i2c reads and writes. 
This way all i2c logic has been moved from MRLComm to Java land. 
Adafruit16CServoDriver can now act as both a ServoController and a MotorController, since it can do true pwm. It's also capable of digtial and analog writes. All this is done using the I2cController interface that is implemented in Arduino / RasPi and Esp8266_01. 
It was working well in some weeks ago. but since then, the GUI and the Adafruit16CServoDriver has been halfway refactored.

MatsMRL, 2017-07-17:
Completed the refactoring of i2c services including the GUI. Closing this issue.

closed 2017-07-17 by MatsMRL