MyRobotLab/myrobotlab #21: the ps3 eye isn't ricognized by myrobotlab

andryspark, 2015-03-29:
i follow this tutorial https://www.youtube.com/watch?v=QkI9AFsBcMk
and when i try to tracking an object so when i run the program, myrobotlab don't ricognize the ps3 eye camera that works well because i tested it  and i can't do anything. I use a laptop with windows 8 and i know that other users will not work ps3 eye camera with myrobotlab too so please fix this bug if it's possible! Sorry for my bad english!

kwatters, 2015-03-29:
Hi AndrySpark,  A lot of times issues like this are due to some library not being found.  The myrobotlab.log file usually contains additional information about what the error was.  If you can, delete the myrobotlab.log file (to clean it out)   , re-run your script, and send a "no worky"   (that will post the log file so that we can review the error message)  (no-worky is in the help menu)

supertick, 2017-08-27:
closing this unless more input or testing is possible.
PS3 has always been a bit of a PITA.  But we could look further if someone can help with testing

supertick, 2017-08-27:
no required for Manticore release - removing from project board

closed 2022-02-06 by supertick