MyRobotLab/myrobotlab #22: MRL should play nice when the computer runs out of disk space

kwatters, 2015-04-18:
MRL will crash and hang if it's doing the "install  all" and runs out of disk space.  We should be more graceful when running out of disk space.

1482 [main] INFO  class org.myrobotlab.service.Agent  - spawning -> [C:\Program Files\Java\jre7\bin\javaw -Djava.library.path=libraries/native;libraries/native/x86.64.windows -Djna.library.path=libraries/native -cp ./;./myrobotlab.jar;./libraries/jar/* org.myrobotlab.service.Runtime -service gui GUIService python Python -fromAgent ]
1482 [main] INFO  class org.myrobotlab.service.Agent  - Agent finished spawn 2015.04.18:10:47:14
13080 [runtime_9244] ERROR   - leaving StreamGobbler
13080 [runtime_9244] ERROR class org.myrobotlab.logging.Logging  - ------
java.io.IOException: There is not enough space on the disk
        at java.io.FileOutputStream.writeBytes(Native Method)
        at java.io.FileOutputStream.write(Unknown Source)
        at org.myrobotlab.framework.StreamGobbler.run(StreamGobbler.java:45)
------

hairygael, 2015-04-18:
I did the install all on my windows 7 with succes.
But I see this version has systematicaly zombies remaîning after shut down which wasńt the case with version 107

Gael

supertick, 2015-04-19:
"Agents" - not zombies ...
The design it a little different - theoretically in the new system
"zombies" will not exist.  But there has to be one service to control all
other mrl instances.  That is what an Agent is.  I have begun to work on
details and documentation of how to use Agents

Roughly,
you start MRL and it starts an Agent.
The Agent does not have a GUI - (it could, but it does not by default)
It turns around and creates a new MRL instance with all the appropriate
environment variables, parameters, then it automatically connects std:in,
std:out, & std:error
Although its a bit rough now.. this will be very important when multiple
instances of mrl are commonly attached.

Its one of my priorities to work on the service page documentation for
Agent & CLI .. it will become more clear when this is done.
Thanks for the feedback !

hairygael, 2015-04-19:
Hi Greg,
Thanks for the clarification about the Agents.
So when an Agent of java.exe remains in the "Task manager" after shutting
down MRL, it shouldn't cause problems like it use to do before?
On another topic, I keep burning jaw servos... I reprinted the whole jaw
and face after coming back from India, because the head had been smashed in
the plane. So when rebuilding it I made sure the servo was moving very
freely with no constraints. Well since then, I burned two more servos...
I know you are super busy, but would you mind checking what could be
causing that between my script and the code behind the speech for InMoov
jaw? I believe it is something with the initialization of the jaw, maybe my
script has something not compatible withth back code about the min max, or
setmouth...

If you check my script, I have also added two motors for the wheels and
direction, could that be implemented as i01.? the two servos are connected
to the 12 and 13 pin of the right Arduino Mega, which correspond to the
rothead and neck.

I have been trying few things with ProgramAB but I will need to go deeper
to really understand the process... Maybe when I see kevin in Paris, he
will give me some lessons, ehehe.

My wife and I are trying to go to San Francisco Makerfaire which is super
huge! It would be great if we could meet again over there!

Gael

supertick, 2015-04-21:
Hi Gael,

>> So when an Agent of java.exe remains in the "Task manager" after shutting
>> down MRL, it shouldn't cause problems like it use to do before?

Correct, Agents do not open ports which is the resource contention with zombies.  Agent's use a small amount of memory - but their purpose is to prepare a correct environment for a MRL instance.  In the future they should be able to monitor other instances, and provide an interface which you can see all mrl instances - imagine a task manager dedicated to watch, monitor and control MRL instances..  Right now it cannot identify zombies and it would be fine if you wiped all instances out

A quick command to quickly wipe out all javaw.exe's (rather than clicking through the task manager or relying on the agent to tell you) is this:

taskkill /f /im javaw.exe

kills all java with one command...

MouthControl should be fixed - the methods names need to change too, they are not standard for Java - and this will break scripts :( but its necessary.  Kevin mentioned in the shoutbox how he saw something amiss regarding your script and how MouthControl behaves... I'll enter an issue for this so we can track it.

San Fransicso sounds fantastic ! It would be great to meet again !  I'm excited your wife might get to go !

supertick, 2015-04-21:
Sorry for hijacking your post kwatters :)
Froze? What parts froze ?  The GUI ? That would be surprising..

kwatters, 2015-04-21:
No worries Grog, all good stuff in the comments here.

The repo dialog box just hung saying it was continuing to download and install stuff,   I think we need to capture the IOException for the out of disk space , pop up a dialog box (or similar) to prompt the user to clean up disk space, and then re-try the download.   

Not sure how far we want to go with this.  I just wanted to track it as a way to harden MRL.  (especially when installing on small embedded devices like the ras pi , or other limited storage systems.