MyRobotLab/myrobotlab #23: MouthControl Refactored

supertick, 2015-04-21:
MouthControl will burn servos - probably it has to do with setmouth and servo limits...
I'm guessing limits are being reset or set incorrectly somehow.

Also .. method names do not follow Java convention..

johnharland, 2015-04-21:
I have a head on Azul and Scooter and have been lucky to not burn out any jaw servos. I have burned servos out on the eyes and now don't have the eye servos in place. 
I am using the following limits for scooter: 
jaw.setMinMax(50, 110)
i01.head.jaw.map(0,180,110,50)      (note the reversal)
i01.mouthControl.setmouth(50,110)
Which follows what Greg has said above and scooter has been on for hours.
I now have the PIR sensor working and want to leave scooter on and "plugged in" to the base all the time. I want it to wake up when I come in the room and respond.
John

kwatters, 2015-04-21:
Hey Guys,  I briefly chatted with Gael about this.  Not sure why he keeps burning out those servos.  I agree, it's likely a calibration issue, maybe some issue with the gearing?  or maybe a bad servo?  

One thing I was thinking, we could have the servo detach when it's not speaking, and re-attach when it is speaking...  That might be a pretty simple fix to add some longevity to the servos...

hairygael, 2015-04-21:
@ Kevin,
Between 12 to 13 servos burned with the jaw, using different brands and
types. I have recently reprinted the head and made sure all parts were
correctly aligned with nothing forcing on the servo. I think mecanicaly it
is safe for the servo.
My settings for the jaw servo are as follow:
i01.head.jaw.setMinMax(60,90)
#i01.head.jaw.map(0,180,60,90)
i01.mouthControl.setmouth(60,90)

I have replaced the servo again this morning because a television was
coming for an interview...

The detach and re-attach option could be a good solution to, at least
preserve the servo longer.

@John, Kevin,
Is your jaw closing completely or is it always a bit open? Mine is set to
be well closed, which of course, forces the servo a little, but once the
mouth is closed, it doesn't buzz anymore.

johnharland, 2015-04-22:
Gael I don't have either robot have its jaw close completely.  They buzz if
closed tightly. The 2nd head for scooter is much better fit then the Azuls.
I printed all the parts on a Fortus 400 which comes out nicer than my
MakerBot. Parts are flatter and no warping.
So maybe not having the jaw close all the way is the reason my servos have
held up. I have scooter talk when ever it does something and am using an
Australia kids voice.

jhack000, 2015-04-23:
Sorry to hear all the problems with this service, I taught myself how to program and so this service  is not the best, but i don't know if it is killing servos but let's do something about . Since this service need some work, I was thinking we sould add a function that only attaches the mouth when inMoov speed and unattaches when inmoov finishes speeking,

hairygael, 2015-04-27:
Hi Jhack, there is nothing to be sorry about!
The jaw service is great, in fact when the jaw isn't working the robot is
definitly missing personality!
So indeed, maybe a detach and attach only when the jaw speaks is maybe the
way to go.
Can we test that easily?

Gael

kwatters, 2015-04-27:
I just committed some code that you should be able to specify

MouthControl.autoAttach = True
or
MouthControl.autoAttach = False

If the mouth is set to auto attach=true, it will only leave the jaw servo attached while it is speaking.

hairygael, 2015-04-27:
Woouii great!
That will only work with version 116 correct?

kwatters, 2015-04-28:
yup. should be in 116. (not released yet..)  I suspect we'll try and fix a few more bugs before releasing 116.

Calamity01, 2017-04-30:
I have replace the internal MouthControl autoAttach method by the autoEnable/Disable method of the servo service to avoid confusion betweens the two methods.

By default, mouthcontrol will enable autoEnable/Disable of the servo. But you can turn that off with the method mouthcontrol.enableAutoAttach(True|False)

hairygael, 2017-05-01:
@calamity, great!
I downloaded version 1.0.2132, is it baked in?

Calamity01, 2017-05-01:
yes it is

hairygael, 2017-05-01:
okay, thank you. I will test tomorrow tha jaw movement

hairygael, 2017-05-01:
Okay, I just tested and the jaw is still moving very strange, it's like the servo is inverted...
For info the head is now moving very slow neck and rotHead. The sliders do not trigger the enableAutoEnable either for neck, rotHead, rollneck. Slider for jaw works.

Calamity, 2017-05-02:
Any chance of taking the right person next time? I've never heard of this before... ;-)

Calamity, 2017-05-02:
Tagging, I mean!

Calamity01, 2017-05-02:
Look like there is a calamity confusion LOL

I will look at it

moz4r, 2017-05-02:
hey ! nothing related to mouthcontrol just about neck, rotHead, rollneck comment
I disable setAcceleration ( smoother movement ) : was causing strange issues like too many servo events causing console stuck. Sometime.
Not important , will open an other post an other day.
But I still have sometime too many servo events caused by sliders I think it too quickly sometime activate/desactivate servo 

22:17:19.960 [i01.head.rothead] INFO  c.myrobotlab.framework.Service - i01.head.rothead.detachPin(13)
22:17:20.116 [i01.head.rothead] INFO  c.myrobotlab.framework.Service - i01.head.rothead.attachPin(13)
22:17:20.117 [i01.head.rothead] INFO  c.myrobotlab.framework.Service - servoMoveToMicroseconds servo i01.head.rothead id 3 120.0->1678 us
22:17:20.121 [i01.head.rothead] INFO  c.myrobotlab.framework.Service - i01.head.rothead.detachPin(13)
22:17:20.355 [i01.head.rothead] INFO  c.myrobotlab.framework.Service - i01.head.rothead.attachPin(13)
22:17:20.356 [i01.head.rothead] INFO  c.myrobotlab.framework.Service - servoMoveToMicroseconds servo i01.head.rothead id 3 56.0->1238 us
22:17:20.362 [i01.head.rothead] INFO  c.myrobotlab.framework.Service - i01.head.rothead.detachPin(13)
22:17:20.662 [i01.head.rothead] INFO  c.myrobotlab.framework.Service - i01.head.rothead.attachPin(13)
22:17:20.663 [i01.head.rothead] INFO  c.myrobotlab.framework.Service - servoMoveToMicroseconds servo i01.head.rothead id 3 51.0->1204 us
22:17:20.669 [i01.head.rothead] INFO  c.myrobotlab.framework.Service - i01.head.rothead.detachPin(13)
22:17:20.803 [i01.head.rothead] INFO  c.myrobotlab.framework.Service - i01.head.rothead.attachPin(13)
22:17:20.805 [i01.head.rothead] INFO  c.myrobotlab.framework.Service - servoMoveToMicroseconds servo i01.head.rothead id 3 46.0->1170 us
22:17:20.810 [i01.head.rothead] INFO  c.myrobotlab.framework.Service - i01.head.rothead.detachPin(13)
22:17:21.226 [i01.head.rothead] INFO  c.myrobotlab.framework.Service - i01.head.rothead.attachPin(13)
22:17:21.227 [i01.head.rothead] INFO  c.myrobotlab.framework.Service - servoMoveToMicroseconds servo i01.head.rothead id 3 46.0->1170 us

very random issue

and hi calamity bro :)

Calamity01, 2017-05-02:
setAcceleration is still at an very early stage of implementation. you should not use it yet until it's more mature. I'm not surprised that with slider it cause problem since the slider send many moveTo event and if it reach the position before the next moveTo, it will always run a a slow speed.

The autoEnable/Disable will of course trigger a lot of messaging between the arduino & mrl. In your part of log, all event happen in the right order, The only thing I could see is that it add a delay of 1ms for the enable messages. That should not be a worry.

Of course, if I add a time before disabling the servo, it will probably help with the number of messages send to the arduino

closed 2017-05-06 by moz4r

supertick, 2017-05-06:
Uh ok ....I have not looked at this.  But I trust you guys keeping the blue smoke ;) 
Booyah !