MyRobotLab/myrobotlab #28: Myrobotlab does NOT open

Jehoiada, 2016-01-09:
Hey guys, I have been trying to get the myrobotlab GUI open for weeks now, and no tutorial is helping me enough. I have all files unzipped and installed, but I can't find the GUI executable. Thanks in advance guys.

kwatters, 2016-01-09:
Hi Jehoiada,
 Did you install Java?  you need java installed.  when it's installed,  you should download the myrobotlab.jar file into a directory.  And from a command line run:   "java -jar myrobotlab.jar"   
Hope this helps!

Jehoiada, 2016-01-09:
Ok, I am using the 1.0.107 version of myrobotlab, because I thought the newer version might have some internal issues. I do also have java installed, but it keeps telling me in the command line, "ERROR: unable to access jarfile myrobotlab.jar". Thanks in advance.

kwatters, 2016-01-09:
It sounds like you have a corrupted jar file.  

1. grab the latest build.  You're on an ancient version.
2. make sure you have java version 1.7 or newer.  Older versions will NOT work

Please provide more information.. what operating system are you using?  is it 32 bit? 64 bit?

Jehoiada, 2016-01-09:
I am using Windows 8.1 64bit, and I just installed the 1.0.119 version.

kwatters, 2016-01-09:
Did version 1.0.119 work for you?
1. create a folder called  c:\mrl
2. put the myrobotlab.jar file into the c:\mrl folder.
3. open a command prompt  ( start -> run -> cmd )
4. in the command prompt type in   cd \mrl
5. next start myrobotlab by typing in  "java -jar myrobotlab.jar" 

Let me now if that worked for you..   for more help, i recommend going to www.myrobotlab.org and say hello in the shoutbox.

Jehoiada, 2016-01-09:
ok, good news, we got the GUI open with the instructions you gave me. but It was the 1.0.107 version. The reason for this is my InMoov project, the developer (Gael Laevigne) said that the newer versions would cause trouble with the code that he already wrote. Thank you for your help kwatters!

kwatters, 2016-01-10:
Hi Jahoiada,
  I'm glad it's working for you now!  You should stop by http://www.myrobotlab.com  and say hello!  People are usually around and can help answer questions as you have them!
Best,
-Kevin

closed 2016-01-10 by kwatters