MyRobotLab/myrobotlab #50: How to create a gesuture with aiml

mishiki1002, 2017-04-25:
Hello i have a replica of the inmoov robot and i would like to develope my own gestures as well as response i have read through lots of the code and replicated my own service and created my own bot. but i have not yet been able to figure out how you guys recreate your gestures. I have searched your website and i have not found anything relevant to it if you could point me in the right direction please, thank you

hairygael, 2017-04-27:
Hello,
I assume here that you are using my scripts and MRL version from this
tutorial:
http://inmoov.fr/how-to-start-myrobotlab/

The AIML doesn't contain the gesture itself.(Although it's possible). It
points to the Gesture directoryr which contains each gesture script.
For exemple in the gestures.AIML:
https://github.com/MyRobotLab/pyrobotlab/blob/master/home/hairygael/AIML/Bots/inmoovWebKit/gestures.AIML

You see:

<topic name="GESTURES">
<category><pattern>REST</pattern>
<template><random>
          <li>Ah, relaxing.</li>
          <li>Taking a rest</li>
        </random>
        <oob>
          <mrl>
            <service>python</service>
            <method>exec</method>
            <param>rest()</param>
          </mrl>
        </oob></template>
</category></topic>

This tells to MRL to look for "rest()" in the GESTURES directory:
https://github.com/MyRobotLab/pyrobotlab/tree/master/home/hairygael/GESTURES

And this bellow is the "rest()" python script:

def rest():
  #neopixel.write(1)
  i01.setHandSpeed("left", 1.0, 1.0, 1.0, 1.0, 1.0, 1.0)
  i01.setHandSpeed("right", 1.0, 1.0, 1.0, 1.0, 1.0, 1.0)
  i01.setArmSpeed("left", 1.0, 1.0, 1.0, 1.0)
  i01.setArmSpeed("right", 1.0, 1.0, 1.0, 1.0)
  i01.setHeadSpeed(1.0, 1.0)
  i01.setTorsoSpeed(1.0, 1.0, 1.0)
  i01.moveHead(90,90,90,90,43)
  i01.moveArm("left",5,90,30,10)
  i01.moveArm("right",5,90,30,12)
  i01.moveHand("left",2,2,2,2,2,90)
  i01.moveHand("right",2,2,2,2,2,90)
  i01.moveTorso(90,90,90)

Hope this helps.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

mishiki1002, 2017-05-20:
Thank you. Sorry for a late response

closed 2017-05-20 by mishiki1002