MyRobotLab/myrobotlab #51: Working Good Sounding Speech

supertick, 2017-04-30:
We need a one clicky, good sounding Speech engine which conforms to the SpeechSynthesis interfaces

juergmaier, 2017-05-02:
Minimal script for using german language. 

Copied the voice jar into libraries/jar folder "voice-dfki-pavoque-neutral-hsmm-5.2.jar"

sent a noworky, script is:

# Start InMoov
i01 = Runtime.createAndStart("i01","InMoov")

######################################################################
# create the speaking service 
######################################################################
i01.mouth = Runtime.createAndStart("mouth", "MarySpeech")
i01.mouth.setVoice(botVoice)

i01.startMouth()
i01.mouth.speakBlocking("Hier spricht Marvin, noch etwas Geduld bitte")

moz4r, 2017-05-02:
Hi juerg this is worky thing in a raw script

```
botVoice = "dfki-pavoque-neutral-hsmm"

# Start InMoov
mouth = Runtime.createAndStart("i01.mouth", "MarySpeech")
i01 = Runtime.createAndStart("i01","InMoov")

######################################################################
# create the speaking service 
######################################################################
i01.startMouth()
i01.mouth.setVoice(botVoice)
i01.mouth.speakBlocking("Hier spricht Marvin, noch etwas Geduld bitte")
```

in full repo script just change those values without downloading any jar
MyvoiceTTS=MarySpeech
MyvoiceType=dfki-pavoque-neutral-hsmm
MyLanguage=de

LunDev, 2017-05-03:
@juergmaier you're still using NaturalReader in the background, to use MarySpeech with the InMoov service something like this would be required:
```
botVoice = "dfki-pavoque-neutral-hsmm"

#create InMoov
i01 = Runtime.create("i01","InMoov")

#switch speech to MarySpeech
python.reserveRoot("i01.mouth","MarySpeech", "MarySpeech", "better speech")

#start
i01.startMouth()
#i01.startAll()

#set CustomVoice
i01.mouth.setVoice(botVoice)

#speak!
i01.mouth.speakBlocking("Hier spricht Marvin, noch etwas Geduld bitte")
```

ore use moz4r's suggestion : )
But note the difference betweeen "mouth" and "i01.mouth".

Another quick tip to macke your life easier, you cann call
`i01.mouth.installComponentsAcceptLicense(botVoice)`
to install a voice auto-magically (instead of downloading it manually).
Just make sure to restart.

Good luck,
Marvin : )

juergmaier, 2017-05-03:
Thanks LunDev and moz4r
My problem looks to be using "mouth" instead of "i01.mouth" in the createAndStart line for the mouth
Now I am happy again, things are worky for me!

closed 2017-05-03 by juergmaier