MyRobotLab/myrobotlab #52: 4 InMoov scripts packaged in the myrobotlab repo - and used as self contained examples

supertick, 2017-04-30:
To make it easy for noobies (and rewarding)  - we want 4 scripts which can be packaged with mrl - and user can easily select one of these - and execute - and it will create virtual arduinos & virtual inmoov

hairygael, 2017-04-30:
my original inmoov needed some tweaking which resulted having the shoulders not with same angles and so on.
I wonder if having gestures in the inmoov.java is still a good idea, it's confusing because some users are searching for those gestures in the folders and can't find them.
def rest():
  i01.setHeadSpeed(1.0, 1.0)
  i01.setHandSpeed("left", 1.0, 1.0, 1.0, 1.0, 1.0, 1.0)
  i01.setHandSpeed("right", 1.0, 1.0, 1.0, 1.0, 1.0, 1.0)
  i01.setArmSpeed("left", 1.0, 1.0, 1.0, 1.0)
  i01.setArmSpeed("right", 1.0, 1.0, 1.0, 1.0)
  i01.torsoSpeed(1.0, 1.0, 1.0)
  i01.moveHead(90,90,90,90,90,90)
  i01.moveArm("left",5,90,30,10)
  i01.moveArm("right",5,90,30,10)
  i01.moveHand("left",2,2,2,2,2,90)
  i01.moveHand("right",2,2,2,2,2,90)
  i01.moveTorso(90,90,90)

def daVinci():
  i01.setHandSpeed("left", 1.0, 1.0, 1.0, 1.0, 1.0, 0.65)
  i01.setHandSpeed("right", 1.0, 1.0, 1.0, 1.0, 1.0, 0.65)
  i01.setArmSpeed("left", 0.75, 0.75, 0.75, 0.75)
  i01.setArmSpeed("right", 0.75, 0.75, 0.75, 0.75)
  i01.setHeadSpeed(0.75, 0.75)
  i01.moveHead(80,90,90)
  i01.moveArm("left",0,118,29,74)
  i01.moveArm("right",0,118,29,74)
  i01.moveHand("left",50,28,30,10,10,47)
  i01.moveHand("right",50,28,30,10,10,137)

def armsFront():
  i01.moveHead(90,90,90)
  i01.moveArm("left",10,115,100,50)
  i01.moveArm("right",10,115,100,50)
  i01.moveHand("left",60,0,0,0,0,0)
  i01.moveHand("right",60,0,0,0,0,180)
  i01.moveTorso(90,90,90)

We might need one more which would involve the head movement. Like a head movement demo.

Calamity01, 2017-04-30:
My view on this is it currently usuless to have the hardcoded gesture. As
you said it make it hard for newbie to find them and prevent user to
customized the animation.

But if we want to do a one click button to start inMoov (scriptless default
inMoov, basic stuff for quick demo), then it can be useful to have some
hardcoded gesture.

So there is pro and con to keep them or remove them

supertick, 2017-04-30:
Removed hardcoded gestures - armsFront, armsUp, daVinci, figher, fistHips, giving, hello, lookAtThis & victory from InMoov service

supertick, 2017-04-30:
What about - handClose and handOpen - don't people have different sets of values for these ?

kwatters, 2017-05-02:
Removing the gestures from the InMoov service is out of scope for this issue.  I've added the 5 default minimal scripts as a unit test, so we validate that the syntax for those doesn't break.  I'm going to close this issue because this is now done.

closed 2017-05-02 by kwatters