MyRobotLab/myrobotlab #53: OpenCV crashes, filters have problems

supertick, 2017-04-30:
OpenCV needs a lot of fixing up.  There are many issues with this service.  It needs auto-testing and we should also promote some interfaces for finding faces and other high level tasks

ShaunHolt, 2017-04-30:
Toshiba Satellite Laptop, Win 10, MRL 2125... will post my note for this here....

Installed version 2125 of MRL, Started up, installed OpenCV service, restarted MRL.

Started OpenCV service, Sarxos grabber. Fps average between 15 and 20 at 640x480.

PyDown filter added, res is 320 x 240... running fine.

Add FaceDetect filter, camera feed freezes and camera disconnects, Do a stop on capture and start again, begins working fine.

FPS seems to average around 15-18 fps

Will let run for about 15 to 20 mins then cont. notes.

ShaunHolt, 2017-04-30:
When capture is not actively detecting a face, fps drops to about 8.
When face is detected fps jumps back to about 15 fps.

Adding Gray filter...

MRL freezes.

Restarted MRL, started opencv service... added pydown, then facedetect, then gray, then clicked capture... nothing happens.

Restarted MRL, started opencv service... added pydown, then gray... works fine.
Added facedetect, and fps drops low, gray and pydown seem to be cancelled out.

Noticed while changing filters and experimenint, the input box for the filters to choose for each filter, continues to grow, possibly causing issues.

It also appears that there is something about the Gray filter that is causing issues. Will continue testing.

supertick, 2017-05-01:
Sarxos frame grabber tested on GroG branch on x86.64.linux (Ubuntu) .. very worky - speed comparable to OpenCV frame grabber

supertick, 2017-05-01:
need to :
- have text field for file capture
- verify frame grabber in input filter is the one used
- figure out why input of filters is not selected - and possibly not used
- input & {name}.input are both entered

juergmaier, 2017-05-01:
2132, only opencv service started using swinggui runtime, no other services running
video input selected, picture shown, 10 fps (640*480) (w10/64, java 1.08, i7 7500U, Intel 620 graphics)
adding affine filter -> worky
adding gray filter -> worky
adding face detect -> NO worky, 
stop/restart capture unfreezes the preview
with activated faceDetect fps vary a lot, 4 .. 14 fps seen, faces are recognized

juergmaier, 2017-05-02:
I get different behaviour when trying to set filters in opencv. Sometimes it works as expected, sometimes filters can not be added, sometimes frame rate degrades to 1 fps - sometimes gui freezes. 
[trackHumans.zip](https://github.com/MyRobotLab/myrobotlab/files/971077/trackHumans.zip)

juergmaier, 2017-05-04:
version 2141:
running only opencv and head in swing gui.
I have a transpose, gray, pyramid down and face detect filter set by script.

Issue 1:
The preview does not take these filters into account. Activating one of the filters in the list earlier did show the output of that filter. This currently works only on my transpose filter. Selecting one of the other filters always shows the full color picture without the previous filters applied.

Issue 2:
Selecting, Adding or removing a filter rearranges the window - can't see a reason why this is necessary?

Issue 3:
faces do not get detected - maybe a follow up of issue 1 - the input picture for the face detect is not filtered?

++ I can remove and add filters, no freezing detected so far
-- performance
my cam has 640*480 pix on USB 2 port, running on w10/64, i7 U7500 CPU, Intel 620 graphics, 
my VideoInput framerate with all filters in place dropped to 1 fps 
switching the grabber to Sarxos does not improve the framerate
having only transpose and gray filter in place results in 2 fps
with transpose only filter fps look to speed up but the swing gui does not show the rate in pic portrait form.
without any filter and VideoInput I get steady 10..12 fps, no dropouts
without any filter and Sarxos I get 9 .. 13, sometimes 20 or higher is shown but varies a lot

supertick, 2017-07-15:
Thanks for the detailed reports.
The filter insertion and removal in the current OpenCV service is "very flawed"...
We are in process of re-writing a new Vision service.  It has a very different filter mechanism, and won't be plagued by many of the problems OpenCV currently suffers.
However, its just begun development so it will take time to make it as capable of the OpenCV service without its problems.

Closing this ticket as we have an future goal now - 
fatal issues in OpenCV will possibly be addressed, but anything with lower priority will be addressed in future development of the Vision service.

supertick, 2017-07-15:
Skeletal Vision service checked in

closed 2017-07-15 by supertick