MyRobotLab/myrobotlab #55: AutoDisable disable servo before the end of the movement

Calamity01, 2017-04-30:
AutoDisable disable servo before the end of the movement.

The issue is cause by a velocity setting not close to the actual speed of the servo.

It could be a bit hard to found the right velocity setting for each servo. Especially since the maximum velocity a servo can go can vary greatly between inMoov build.

One idea to fix this is to add a grace periode before disabling the servo. So it disable after 1 sec or so after arduino think it finish it's movement

What should be the grace period? 1 sec, 0.5s, 5 sec?

moz4r, 2017-04-30:
I like grace period, 5seconds will be good. if servo electrized more better as not enough I think

Calamity01, 2017-05-12:
add the grace period and set it to 1s. That should be enough to take care of slight variation on the velocity because of load or low battery.

I did not want to set it so it will give enough time to move even if the velocity setting is set wrongly. In the long term, it will cause more problem than that solve. We should try to use a velocity value that is close to the actual value the servo move

hairygael, 2017-05-12:
Thanks Calamity,
Will this fix also the jumpy effect when using the sliders? I do not know
if you noticed, but enableAutoDisable added to enableAutoEnable , makes the
sliders very inaccurate.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2017-06-20:
I think it would be more useful if gestures had a start & finish event, rather than try to calculate, or guess time.

I added i01.startedGesture() & i01.finishedGesture() to InMoov service .. in the script they can be used to box in a gesture.  When called 
startedGesture:
         RobotCanMoveRandom = false;
         enableAutoDisable(false);
finishedGesture:
     RobotCanMoveRandom = true;
     enableAutoDisable(true);

so in theory it should be allowing gestures to complete...
Gael last reported that - it does not appear that auto disable feature is active again when the gesture completes .. this could be because of details of that feature ...

moz4r, 2017-06-20:
About history, I have added forceElectrize(X), X was estimated time the gesture will run ( no need an accurate value, just the higher possible ) . forceElectrize just increase default disable timer ( 10 seconds if velocity = -1 / moveTo delay + 1 second, if velocity > -1 )

Ok to simplify we can use i01.startedGesture() & i01.finishedGesture()  but we need to put i01.disable() inside i01.finishedGesture() to force disable at the end . enableAutoDisable(true) is not enough.
This will work only if servo is not moving, because a not finished moveTo will REelectrize the servo. 
Maybe we can put a new thing like i01.stopMoving() Inside i01.disable() 

All servo will disabled. The risk is to shutdown servo too early at the end of the gesture. So put a sleep just before i01.finishedGesture() .
Second risk is enableAutoDisable(true) will override user config value ( exemple if user had set neck servo to never autodisable Inside config )

To simplify again we can set autodisable a new default value for all, because i01.finishedGesture() will override users values.

To look further a moveTo blocking action will help us I think

moz4r, 2017-08-01:
- slider jumpy effect fixed  
- i01.startedGesture() & i01.finishedGesture() fixed

[image]

Those 2 commands will not override user values about default autodetach, that is good
Sometime a sleep(x) is welcome just before i01.finishedGesture() , like at the end davinci

closed 2017-10-02 by moz4r