MyRobotLab/myrobotlab #62: http://localhost:8888/#/service/i01.ear

hairygael, 2017-05-01:
I don't seem to be the only one having this issue, Markus had the same today.
http://localhost:8888/#/service/i01.ear doesn't launch the microphone on Google Chrome when using version 1.0.2132. The webgui is correctly launched though.
We do not launch the whole webgui but only the ear.
This is happening when launching the inmoov repo with START_INMOOV_BAT and also when using the minimalHand.py.
For info, it was worky with version 1.0.2122

moz4r, 2017-05-03:
tested last build  with this little fix on repo script+minimal script, ok for me
better for you ?

kwatters, 2017-05-03:
It's important to make sure you launch the browser after the i01.ear service has been started.  Otherwise, you'll get a blank browser page.

hairygael, 2017-05-03:
Thanks for this important info, I modified the minimalHand script on the
InMoov repo.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2017-05-04:
Has this been verified worky ?
Can it be closed ?

hairygael, 2017-05-04:
yes worky!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

closed 2017-05-05 by hairygael