MyRobotLab/myrobotlab #67: Create MarySpeech GUI similar to its online webdemo - to switch voice and effects

supertick, 2017-05-03:


hairygael, 2018-05-28:
> Cooked !

Hey very nice!!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

johnharland, 2018-05-28:
I tried it in 1.1.6 and it is vey neat. Will be easier to pick a good voice!

hairygael, 2018-05-28:
Quick thought:
Is the volume adjusting the system volume?
If no, it is better to set by default to max.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2018-05-28:
The volume adjust only audiofile utterance related, so it is hardware universal. by default it is full power

[image]

hairygael, 2018-05-28:
Great!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

closed 2018-06-07 by moz4r