MyRobotLab/myrobotlab #69: Create WebGuiClient service

supertick, 2017-05-04:
This solves quite a few other issues and prepares the way for large P2P BotNet control

kwatters, 2017-05-04:
webgui client ?  sounds like overlapping functionality for the remote adapeter ?  
Just curious what you are thinking/planning here.

supertick, 2017-07-15:
The MRL API - is relatively ubiquitous in its accessibility.  
http://myrobotlab.org/content/myrobotlab-api

It is used by the Web UI, and Cli (now recently refactored into a single modules, which both services share)
Its is also used for REST calls, and potentially a dynamic swagger interface can be created. 
The swagger definition would allow many other languages, instant code generation for interfacing.

Additionally it will resolve the control issues of the Agent service.
I envision when mrl starts it will start the a process with an Agent service (as it does now) - the Agent will spawn the target mrl instance with the appropriate command line parameters (ie - jni, jna, classpath, jvm & memory directives).  The Agent has control of sub-processes, because it has the process handle, but it can not do any form of other control or communication, because it only had stdout stream - and that is crowded with logging and other information.

So, I want the Agent to open its own webgui on port 8887 .. the "default" mrl instance will not start a webgui (only swing), but will try to connect to localport 8887 of the admin.  These two instances will be joined and the Agent's webgui will allow the user to control the all local mrl instances spawned by the Agent.  There is no port contention, because there is only 1 port/web server open on port 8887 .. (the Agent's)

WebGuiClient - has been created, but does not at the moment register or provide all the control / feedback features it needs to, however, it does not need to be done for manticore release .. 

closing ticket

closed 2017-07-15 by supertick