MyRobotLab/myrobotlab #71: vinMoov wrong servo direction

hairygael, 2017-05-04:
VinMoov is great and I really love that stuff!!!
 It will GREATLY help a lot of people to see what issues they might have with their robot calibrations.
But to drive them in the correct direction, vinmoov has to be correct. Here are the things that should be corrected:
-vinMoov jaw is moving opposite than real InMoov (but maybe this is related to the current issue with mouthControl. Real InMoov Jaw doesn't work properly)
-vinMoov rotHead turns opposite direction than the real InMoov.
-vinMoov wrists are turning opposite direction than real InMoov.
-vinMoov lowStomach is turning much more than real InMoov. (hard to define, but it's almost twice as much)
My version of MRL is 1.0.2142, using InMoov repo.

Calamity01, 2017-05-06:
there is a way to set the angles of the vinMoov part to fit better the
actual angles of real inMoov.

InMoov.vinMoovApp.setMinMaxAngles(String partName, double min, double max)
where min and max are 'real' angle the vinMoov part will take when servo is
at min or max.
the partname mapping to servo can be found there
https://github.com/MyRobotLab/myrobotlab/blob/develop/src/org/myrobotlab/service/InMoov.java#L1790
I know that did not follow InMoov convention, and want to change the
partName, but never had the chance.

The default parameters I use correspond to what I have for my InMoov, but I
use a lot of hs805 and hk15388 so the mapping between real angle and servo
MinMax may not correspond to a 'standard' InMoov build

By example, for my midTorso servo, I can turn it to about 75 degrees on
each side (and that's how the parameters for vinMoov are set) I know it is
more than what you specify in the BOM page, but I have no idea what angle
your inMoov actual can turn is torso.

hairygael, 2017-05-17:
From Calamity on post MyRobotLab/inmoov/issues/54#

### _no, that line is just a linker so I could make your way of mapping servo to work with vinMoov. I eventually want to remove that line

Better work on how the joint is define
a few line before that line, you will see

rotationMask = ... the value -1 or 1 define the orientation of the rotation (clockwise or counterclockwise)
angle = define the initial position of the part (rotation of the 3d model in his local world)

playing with those two value is better in the long term_
###################

Okay thanks that makes sense.
Has it been modified already?
node = new Node("RWrist");
rotationMask = Vector3f.UNIT_Y.mult(-1);

node = new Node("LWrist");
rotationMask = Vector3f.UNIT_Y.mult(1);

hairygael, 2017-06-17:
node = new Node("RWrist");
rotationMask = Vector3f.UNIT_Y.mult(-1);

So I tried to modify (-1) to (1) but that is not what I desire as result. This affects the initial angle, which means the right wrist is opposite positionned at start of VinMoov. 
What I try to achieve is that when it rotates, instead of going anti clock it goes clock wise.

supertick, 2017-10-20:
I need more information.  
First, what script are we talking about ?  If we are talking about the "service" script, which i think would be appropriate, currently, it does not even start the virtual InMoov.

So, lets start with that - should the very important, official, auto-tested, **Service** Script (https://raw.githubusercontent.com/MyRobotLab/pyrobotlab/develop/service/InMoov.py) automatically start virtual InMoov ... my vote yes...

If we all agree on that, then we can discuss what else should automatically start, or details of progressively starting other parts.

After we start from the same place, we can discuss and easily observe the same problems.  
We will all be on the "same page".

moz4r, 2017-10-21:
I think we are talking about full script from inmoov repo. who just launches lot of standardized mrl core services / optional skeleton parts / and virtual inmoov by default. Because that some people use. 
Inmoov full script is just "start all mrl" and keep personal config safe. MRL core services are the basics. 
Write a second full script that launch mrl core services, why not, we all love write code. If it is needed we will do the job.
But inmoov repo is this one that needs to be auto tested and set as service script ? 
Other subject, inmoov service really need to evolve after manticore release, we definitely need to use shared core for lot of customized hardware declensions / remote adapter etc.. . https://github.com/MyRobotLab/myrobotlab/issues/101
And keep default hardware configuration as main option to make things easier for new people.
Write multiple same things is a nonsense and time waster ( except for learn things, it is great ) , customize is needed. this will go through a revamp of the inmoov swinggui / other things . 
That is just my point of view.
Next release after Manticore will be a great release :)
At this time is wrist still use the wrong direction ? If yes we will fix it asap.

moz4r, 2017-10-23:
Service script Inmoov.py updated to launch virtual Inmoov

hairygael, 2017-10-24:
Hello Greg,
Currently there is no more full script for InMoov. Not even in the
inmoov-develop repo. (The only full scripts are on my pyrobotlab page
<https://github.com/MyRobotLab/pyrobotlab/tree/master/home/hairygael>, but
I do not link or suggest them anymore to builders)
In the inmoov-develop, we have a set of script (which I hope still work) to
launch each part of the robot, finger starter, hand, head, arm, torso.
These scripts are usefull when building InMoov and also for understanding
how a simple script works and what it does.
https://github.com/MyRobotLab/inmoov/tree/develop/InMoov/testScripts

With the inmoov-develop, using the config files, it makes it easier for
most newbie to launch various parts or the full robot, without having a
script of 20 miles long.
With the START-INMOOV.bat, VinMoov is launched by default.

As Anthony suggest, in the future we will keep a default hardware
configuration as main option.
But we also need to give the opportunity to play with other components
which have services available in MyRobotLab.

That will be after Manticore.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

hairygael, 2017-10-24:
Regarding the wrist, I had modified something in java land for to fix it,
but I haven't downloaded the pushed version so I have no idea if it turns
in the correct direction.
https://github.com/MyRobotLab/myrobotlab/commit/e01a0fe685d7d3c22ec63ccbc0dda0e0898e6ba7

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

2017-10-24 13:43 GMT+02:00 gael langevin <hairygael@gmail.com>:

supertick, 2017-10-24:
Thanks Anthony & Gael,
After manticore is released, develop branch will very quickly change.  One of the largest changes on the horizon, is myrobotlab will use "maven" to build instead of "ant", and this will allow us to remove most of the "repo".  After this is working, upgrading components such as opencv, javacv, & openni which have previously been so difficult will be as easy as changing a number in a text file (goodtimes). 

While this upcoming change is very desired it means a lot of fundamental changes to mrl.  What do we have verify things work the same as before with all these changes ?  **Auto-magic elves and service scripts !**  This is the only real way of preserving stability and workiness with such limited resources.

So, this is the reason why I'm interested in service scripts testing all the parts we want to keep stable.  It's the automated tests which will keep our ship stable as we roll into uncharted areas.  Anything we add to the service scripts is **tested every build**.  

There is a lot of great ideas & work in the batch files & "building" scripts.  Perhaps more parts of those can be incorporated into the service scripts too.  We all want worky, and if its tested all the time the more likely it will be worky.

@moz4r - thanks virtual inmoov starts now in the service script.    ...goodtimes...

moz4r, 2017-10-24:
Understand and agree with the roadmap ! 

I can set inmoov repo script "Auto-magic elves friendly" and call it from Inmoov service script ( full ). 
Good / Bad  idea ?

In all cases it will be very useful to auto test it...

Other thing about duplicates,
I think we don’t need that anymore: 
https://github.com/MyRobotLab/inmoov/tree/develop/InMoov/testScripts

But we can use this unique reference place after fusion :
https://github.com/MyRobotLab/pyrobotlab/blob/master/service/InMoovHand.py
https://github.com/MyRobotLab/pyrobotlab/blob/master/service/InMoovHead.py
https://github.com/MyRobotLab/pyrobotlab/blob/master/service/InMoovArm.py
https://github.com/MyRobotLab/pyrobotlab/blob/master/service/InMoovTorso.py
https://github.com/MyRobotLab/pyrobotlab/blob/master/service/InMoovEyelids.py

Because that is used for documentations

supertick, 2017-10-24:
> I can set inmoov repo script "Auto-magic elves friendly" and call it from Inmoov service script ( full ).
> Good / Bad idea ?
> 

Sorry, I don't understand moz4r ... auto-testing elves are very rigid but reliable.  They will look at "all" the services, and expect a pyrobotlab/service/{serviceName}.py to test.  If stuffs can be virtualized, potentially quite a lot of stuffs can be tested within the script.  For example, I have found some services explode on serialization when they run and "do something useful" .. which is different from when the script only starts them.  Testing a service in use is much better than just testing the starting of a service.

Copies of scripts are not as desirable and normalized code, because at some point "one of the copies" will be wrong.  I'm not interested in "configurable" auto-elves to test scripts outside of /service/ directory, because I got it setup where its all automatic (the price to pay for rigid yet reliable auto-testers).

The pyrobotlab/service/scripts are (or will be) hooked up to "examples" .. so documentation in the scripts is very useful, even if auto-elves don't read it.

Calamity01, 2017-10-24:
Gael

Rotation mask regulate the rotation and angle is the initial angle the
piece will take at startup in that way of rotation.

In another word. In your example. You said turn -90 clockwise at startup.
In you change only the rotation. You end up starting at -90 anticlockwise.
Thats why the wrist is turn on the wroung way

Christian

supertick, 2021-08-08:
This is a very old ticket ...
Closing unless there is some servo currently in "real" inmoov that is backwards 
I cannot tell, so others with real InMoov would need to verify.

Closing if no activity....

closed 2022-02-06 by supertick