MyRobotLab/myrobotlab #72: Regarding setVelocity odd behavior.

hairygael, 2017-05-08:
I am trying to set a simple script for beginners using the FingerStarter with setVelocity instead of setSpeed.
But I encounter odd behaviors.
-Servo takes the velocity specified in the def, but if the moveTo is not completely finshed, the next command will finish it first at full speed before starting the specified def .
-Servo doesn't always disable although the def says to disable. (Analog TowerPro MG946R)

Here are the def I'm using:

def fingeropen():
  i01.rightHand.index.setVelocity(20)## Low velocity
  i01.moveHand("right",0,0,0,0,0,0)## Thumb,index,majeure,ringfinger,pinky,wrist
  i01.mouth.speak("ok I open my finger")
  sleep(6)## We give a delay before to disable the servo to finish the movement
  i01.disable()## We disconnect the servo

def fingerclose():
  i01.rightHand.index.setVelocity(50)## Medium velocity
  i01.moveHand("right",180,180,180,180,180,90)
  i01.mouth.speak("my finger is closed")
  sleep(3)
  i01.disable()

def fingermiddle():
  i01.rightHand.index.setVelocity(-1)## Maximum velocity
  i01.moveHand("right",90,90,90,90,90,90)
  i01.mouth.speak("ok you have my attention")
  sleep(3)
  i01.disable()

Calamity01, 2017-05-08:
Hi Gael

setVelocity is acting the same way as setSpeed, there is no difference.
when you do setSpeed(), it convert the Speed unit in a velocity unit (base
on the formula post in another issue) then call setVelocity.

The problem you describe seem relate to autoEnable. When a servo get
Enable, it will move to it's target position at full speed. That's how
arduino make things and we have no way to go around that. But the thing is
that a servo should not get enabled if it's already moving.

So base on what you describe, I think that the Servo service somewhat get
bork on how it remember if he is enable or disable. I will have a look at
that

hairygael, 2017-05-09:
Thanks Christian,
I have another question, what is the reason for to setVelocity specified in the InMoov services.
For exemple the head have a setVelocity(45.0,45.0,45.0,45.0,45.0,45.0);
Which makes the jaw and eyes not functional on my side unless my script overides with a 
setVelocity(-1,-1,-1,-1,-1,-1);

moz4r, 2017-05-09:
setVelocity(45.0,45.0,45.0,45.0,45.0,45.0) is default speed. It is good to set a default velocity instead of max velocity that can break skeleton parts . So we overide fullspeed only if we need it. But default jaw and eyes velocity can be set to default -1 , why not.

hairygael, 2017-05-09:
I understand, it took me a while to get that point.
I couldn't figure why the servos were moving so slow until I found in the
services those default velocity.
Eyes and jaw should be set to (-1) in order to initiate the mouth correctly
when launching InMoov with a minimal script.
Does those default velocity get back on again when we disable and enable
again? Or does it keep the previous setVelocity like setSpeed use to do?

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-05-09:
Pushed tonight if i go home . Car breakdown ;) disable action dont delete speed info. Exept what Christian said

hairygael, 2017-05-09:
Who likes car breakdown?
Thanks for the push and info!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

closed 2017-06-15 by moz4r