MyRobotLab/myrobotlab #75: openNI, shows no image, only depth. My cam is working for both with Processing.

Calamity01, 2017-05-11:


Calamity01, 2017-05-11:
The OpenNi service can manage rbg images, but currently it's not enable as it is not needed for the usage it have been used (skeleton tracking). 

try OpenNi.enableRBG(true) and the getData() method will probably return the image (I did not test if it work)

juergmaier, 2017-05-12:
it's more an issue with the gui. it offers an image butten but does not show anything. I first thought my cam was broken.

juergmaier, 2017-05-12:
do you know how I can access the depth data im my own python code?
e.g. 
openNI = Runtime.createAndStart("openNI","openNI")

def myDepthView:
  openNI.getData()
  ?? but how can I access the data.depthMap array?

Calamity01, 2017-05-12:
when OpenNi is active, it will internally call getData() so you don't have
to use that method

getData will publishOpenNIData so you need to subscribe to that method.

the data returned on the publish event is a data class. You can see the
detail here
https://github.com/MyRobotLab/myrobotlab/blob/master/src/org/myrobotlab/openni/OpenNiData.java
some data may be null if they are not computed

supertick, 2017-08-27:
not required for Manticore release - dropping from project board