MyRobotLab/myrobotlab #76: aref inside Mrlcomm

moz4r, 2017-05-12:
Small question while I thinking of it, to look towards the future, maybe a generic thing to use differents finger sensors, based on autocalibration, analog reference need to be tweeked. Of course we can hardcode aref Inside the sketch before upload mrlcomm. The question is : can we change aref dynamicaly after the upload ? and never touch mrlcomm ?

Calamity01, 2017-05-12:
I think aref state can be set at anytime. So we just need to set a messages
to send to MRLComm set it can set the state for aref. It should be very
basic to set

supertick, 2017-07-06:
The mrlcomm message definition is generated from a schema.
This file contains all message definitions - adding a new message can be done by adding a new line.
https://github.com/MyRobotLab/myrobotlab/blob/develop/src/resource/Arduino/generate/arduinoMsgs.schema

closed 2017-07-06 by supertick

moz4r, 2017-07-25:
pushed mrlcomm update that was fun ! I hope I didn't break anything. Thank for you help guys.
need to polish things because at this time there is no board verification and I send only aref integer value.
( INTERNAL2V56 is better than "3" )
step 2, ask @Ash to lend his finger sensor