MyRobotLab/myrobotlab #77: Freezes on Windows 10

supertick, 2017-05-12:
Humanoid reports,
"Runs once then freezes. I re-opened and used MRL again a few times to move some servos then it froze again and I couldn't close it so I had to restart the computer to get rid of it. 'Close window' from the start bar wouldn't even get rid of the persistent little bugger.  Weird. Hopefully you can sleuth what's happening. "

Hopefully we can get a myrobotlab.log - and/or someone can replicate the problem ?

Calamity01, 2017-05-12:
I have seen that problem from time to time. For me that seem to happen at
the end of loading my InMoov script, Maybe it's relate to the jm3 as it is
the last thing that load.

One more thing. I notice that when it got thru that bug (swing gui working
normally) just after it open the jm3 window, the swing window get "blank"
for like half a second, then all redraw and work fine. When it do not work,
the swing window do not erase, but was freeze and would not work. It's only
the swing gui that are the problem, other MRL function seem to work normally

Humanoid1, 2017-05-12:
I've just sent the log file to Grog. I forgot to mention that when it runs the script once before freezing it moves the servos during that run but doesn't open any services except 1 x arduino service. No servo services start.

supertick, 2017-05-13:
Below is the exception I see in your log file .. but I cannot seem to reproduce it .. it might be related to a race condition perhaps - and the Arduinos taking longer than the virtual Arduinos I used .. I'll look into it more ..

18:35:06.926 [Arduino1] WARN c.m.f.Service [Service.java:1483] no such method Serial.publishPortNames(ArrayList) :  - attempting upcasting
18:35:06.926 [python.interpreter.3] WARN c.m.f.Service [Service.java:1483] no such method Serial.publishPortNames(ArrayList) :  - attempting upcasting
18:35:06.926 [Arduino1] WARN c.m.f.Service [Service.java:1489] searching through 221 methods
18:35:06.926 [python.interpreter.3] WARN c.m.f.Service [Service.java:1489] searching through 221 methods
18:35:06.926 [Arduino1] ERROR c.m.f.Service [Service.java:1512] boom goes method publishPortNames
18:35:06.926 [python.interpreter.3] INFO c.m.s.Runtime [Runtime.java:338] Runtime.createService Arduino2
18:35:06.926 [Arduino1] ERROR c.m.l.Logging [Logging.java:19] ------
java.lang.NullPointerException
	at org.myrobotlab.framework.Service.out(Service.java:1620)
	at org.myrobotlab.framework.Service.invokeOn(Service.java:1506)
	at org.myrobotlab.framework.Service.invoke(Service.java:1403)
	at org.myrobotlab.service.Serial.getPortNames(Serial.java:677)
	at org.myrobotlab.service.Arduino.getPortNames(Arduino.java:2039)

Humanoid1, 2017-05-13:
Thanks,
I would leave the arduinos connected while I reopened the jar file each time mrl froze. (Just in case you can't tell from the log file).

supertick, 2017-07-07:
Found and fixed a bug in the SwingGui.  Other parts of the code are also protected with SwingUtilities function.  If this is an issue with build > 2288 then we can re-open.

closed 2017-07-07 by supertick