MyRobotLab/myrobotlab #78: MarySpeech WebGui Borked

supertick, 2017-05-12:
webgui : ------
 java.lang.NoSuchMethodException: could not find MarySpeech.installSelectedLanguagesAndVoices(ordinal 1) in declared methods

Which is true - the installer has the method, but not MarySpeech .. perhaps it should proxy the method ?

LunDev, 2017-05-12:
no ... the webgui needs to change
that is how it used to be noWorky earlier
Now it's using installComponentsAcceptLicense(String component) & installComponentsAcceptLicense(String[] components)

If it even gets that far

juergmaier, 2017-05-24:
In my current eclipse copy of develop I get an error on MaryUtils.java, toGermanLetterTable line 592. But I do not understand what the function is doing and the character-comments show up as gibberish? Do I need special settings in Eclipse (neon) to see what is going on?

beetlejuiceFR, 2017-05-26:
@juergmaier, This file is located to mrl\src\org.myrobotlab.MarySpeechUtils.java . In Eclipse, clic on it, and in the menu bar, go to file->properties. This will open a new window. In the tab "ressource", you will see " text file encoding" at the bottom. Select "other" and "UTF-8". This should correct the error .

juergmaier, 2017-05-26:
@GroG, worked like you explained, thanks. However still can't figure out what the code does and why I get so many nonstandard characters in the comments?

supertick, 2017-07-06:
Appears to be working in the webgui (original issue)

closed 2017-07-06 by supertick