MyRobotLab/myrobotlab #79: did not find method - onController()

moz4r, 2017-05-19:
Hi !

Trying to kill errors in the black box, I like clean black box ! I dont understand this one

11:50:27.660 [SwingGui] ERROR c.myrobotlab.framework.Service - did not find method - onController()

this caused if I start a servo

An idea ?

supertick, 2017-05-19:
Thanks moz4r .. Ya, I can see the errors.  It has a couple problems.  The controllers nor the control was updating the gui.  Good find !  I'm fixing now

supertick, 2017-05-19:
This was a bit messy.  There was onRefreshControllers and onController and onState.  These are all callbacks which potentially refresh the state of the gui.  onState is the "big" one and its logic was correct.  Sometimes we choose to decompose the methods into more discrete callbacks like onController which would just refresh the Servo's controller state.  In this case it was a bad idea - so I removed it and fixed up onRefreshControllers - there was a multitude of fixes ..

Tested - attaching a controller, detaching a controller,  adding a new controller to the system (it appeared in the dropdown) ..  no ERRORs .. looks good.

closed 2017-05-19 by supertick

supertick, 2017-05-19:
fixed in 1.0.2178

moz4r, 2017-05-19:
Me and the black box thank you a lot grog