MyRobotLab/myrobotlab #81: Color the world

moz4r, 2017-05-24:
Just to talk about default tab/service colors 

[image]

supertick, 2017-05-24:
Looks delicious !   You went much further - I'm curious where your setting this.  In the ancient days the DockableTab did not exist, but colorizing tabs did, it was used exclusively to show "remote services" - when 2 mrls instances attached.  
[image]

I'm all for it ! - the only question is how, where and the details as to what governs the different colors.

moz4r, 2017-05-24:
Hi grog ! I using this after a new tab creation
switch (sw.getClass().getSimpleName()) {
          case "Servo":  tabs.getTabs().setBackgroundAt(tabs.size() - 1,new Color(227,203,195));
and pass a var to DockableTab to store the color.
Final idea is cleaner, using interfaced parameter , and set inside each service this.tabColor=color. if it is a good idea.

supertick, 2017-05-27:
Sorry moz4r ... I wish I had the time to respond quicker.  
I really like your concept, but we will need to change the implementation. 

**MRL IS NOT THE GUI , THE GUI IS NOT MRL**

This is an important axiom. It means that most Services don't need the GUI and the GUI doesn't need other services.  The GUI is just one of many **views** and just another Service of the larger system.  When you merge the GUI with other Services, the requirements of the GUI can affect the rest of the Services.

I'll give you an example - java.awt.Color is for Swing.  The package java.awt.* is just for the GUI. .. Android doesn't have any java.awt.*. So adding java.awt.Color to Runtime prevent MRL from running on Android.

Sometime we need to separate capabilities.  Display capabilities belong in UI Services, other Services have other capabilities. We try to keep these capabilities well defined.  People like to play with clean Lego blocks ... if you glue services together, it gives the end user less options to combine the blocks.

I describe this at length not to criticize, but because I respect and appreciate your contributions, and want you to understand why the Runtime nor any other Service except SwingGui should have java.awt.Color :) 

What I would recommend is a color utility class, which defaults the HSL H value to a hash of the Service name.  The UI Services (Swing & WebGui) could use this utility to color the Services, and if desired the functionality could be added for the user to override the hash values with their own desired colors (to be implemented sometime in the future)

moz4r, 2017-05-27:
Thank Grog for this precious information, so the first idea was better I think ! no need to touch any service.
I will revert and mod only swingGui.java . and set this functionality to optional

moz4r, 2017-05-31:
world colored

closed 2017-05-31 by moz4r

supertick, 2017-06-02:
BooYa - We love color !