MyRobotLab/myrobotlab #82: WebkitSpeech is crazy ( 3 same msg sent )

moz4r, 2017-05-30:
Since last commit about msg ( I think )  webkitspeech send 3 same events at the same time.
Strange 

**09:18:06.678** [New I/O worker #8] INFO c.m.f.Service [WebkitSpeechRecognition.java:126] Recognized : >hello<
**09:18:06.679** [New I/O worker #8] INFO c.m.f.Service [WebkitSpeechRecognition.java:126] Recognized : >hello<
**09:18:06.680** [New I/O worker #8] INFO c.m.f.Service [WebkitSpeechRecognition.java:126] Recognized : >hello<

supertick, 2017-05-30:
Yikes ! Ok, I'll check i out.

supertick, 2017-05-30:
Thanks !
Had to do with refactoring the WebGui - silly mistake - fixed

closed 2017-05-30 by supertick

moz4r, 2017-05-30:
I was stucked on a same mistake some days ago, I didnt know we need to set a break Inside each case condition.
thank again !