MyRobotLab/myrobotlab #83: UTF8 the world

moz4r, 2017-05-31:
UTF8 worky perfect inside every services.
But we need to tweak logging about it because it is hard to debug, if we use UTF8 language, like russian. Especially to debug aiml.
Need help to tweak the console and logging in general

[image]

moz4r, 2017-06-15:
seem worky ! 

[image]  

added to launcher :  

REG ADD HKCU\Console /v CodePage /t REG_DWORD /d 0xfde9 /f
REG ADD HKCU\Console /v FaceName /t REG_SZ /d "Lucida Console" /f
java -Dfile.encoding=UTF-8 -jar myrobotlab.jar

hairygael, 2017-06-16:
Great!
Lets wait for Andrei to tell us.

kwatters, 2017-06-16:
The aim/goal should be for all things to be UTF-8.  I suspect there's still a few places, in particular with ProgramAB that are only ASCII.  As for the console and logging, I think that's a windows OS specific issue and likely has to do with the locale that was chosen when the operating system was installed.

We can probably work at making the file.encoding property default to UTF-8 in MRL if you think that would help.  Clearly, I see that you're able to pass it on the command line.

Worky always wins 👍

AlexInMoov, 2017-06-16:
Antony вот лог файл с новой ошибкой, жалуется на utf8: вот ссылка https://yadi.sk/d/z09zVxzJ3JkPgQ

moz4r, 2017-06-16:
Hi ! thank you for the log file. It say you are using myrobotlab **2105**. 
Please take the last testing one . It include last Inmoov russian chatbot and fixes : **2248**  
https://s3.amazonaws.com/mrl-bucket-02/current/develop/myrobotlab.jar

AlexInMoov, 2017-06-16:
Спасибо Antony заработало но в консоли есть ошибки с кодировкой https://yadi.sk/i/AGJsg93G3KCHzZ

AlexInMoov, 2017-06-16:
Antony подскажите как поменять голос диктора на мужской.

moz4r, 2017-06-16:
@kwatters 1 issue identified about UTF8 encoding
If I use myrobotlab.jar complided by eclipse the console show good characters.
If I use public myrobotlab.jar compiled by travis the console show bad characters like on first screenshot.
An idea ?

moz4r, 2017-06-16:
@AlexInMoov 2 options ( about what I know ) to use a male voice

1/ using amazon poly
```
[TTS]
MyLanguage=ru
MyvoiceTTS=Polly
MyvoiceType=Maxim
AWSAccessKeyId=XXX
AWSSecretKey=XXXYYYZZZ
```

2/ using MarySpeech
There is no builtin voice because it is big file size. you can manualy install the voice : 
- download : http://www.myai.cloud/mrl/mary-mrl-ru.zip
- extract
[image]

- setup

```
[TTS]
MyLanguage=ru
MyvoiceTTS=MarySpeech
MyvoiceType=ac-nsh
```

kwatters, 2017-08-29:
@moz4r , curious, you said "compiled by eclipse" ...  what do you mean?  Are you using eclipse to run ant?  what happens if you build it from the command line with "ant jar"  ?  

Also, do you have any uncommitted changes locally that might be causing a difference?

moz4r, 2017-08-29:
Hi ! I will prepare some test to clarify it

moz4r, 2017-08-29:
Ok, 2 simple test. ( It is not related to programAB but it is the test I used to show you the problem )

**Inside eclipse** service programAB, using magic key "F11" 
All if worky perfect. All is UTF8 friendly
[image]

**Using online public jar compiled by Jenkins** : logging is UTF8 killer
[image]

webgui=Runtime.start("webgui", "WebGui")
ai=Runtime.start("ai", "ProgramAB")
ai.startSession("default", "alice2");
print ai.getResponse("hi there")
print ai.getResponse("こんにちは")

kwatters, 2017-08-29:
Ok, I suspect this is because of the default locale that is set in your terminal window when you run the downloaded myrobotlab.jar.

In eclipse, I suspect it's setting the char encoding to be UTF-8  when it runs.  But when you run from a command prompt, it's not setting this.

There's some info about it here.. maybe this will fix it for you.:
https://stackoverflow.com/questions/388490/unicode-characters-in-windows-command-line-how

Additionally,  I know there was some work at one point to pass the default locale into the jvm with a 
-Dfile.encoding=UTF-8

https://stackoverflow.com/questions/361975/setting-the-default-java-character-encoding

Recently there was a change to pass jvm parameters through the agent to the running myrobotlab instance, so it would have to look something more like:

java -jar myrobotlab.jar -jvmargs="-Dfile.encoding=UTF-8"

(It might not be -jvmargs.. maybe GroG or someone can chime in on the proper syntax to pass this through to the jvm.) for exact syntax

moz4r, 2017-08-30:
myrobotlab.log fixed to be utf8 friendly but there is still a global problem
I think it is identified :

If I launch : 

java -Dfile.encoding=UTF-8 -jar myrobotlab.jar -jvmargs="-Dfile.encoding=UTF-8"

and look about java preferences from myrobotlab.log : 

16:38:58.237 [main] INFO c.m.s.Runtime [Runtime.java:1885] file.encoding [windows-1252]
16:38:58.237 [main] INFO c.m.s.Runtime [Runtime.java:1886] Charset.defaultCharset() [windows-1252]

The both Dfile.encoding are not parsed.

I can try to harcode UTF8 preference inside runtime, is it a good idea ?

moz4r, 2017-11-03:
FaceRecognizer broken if folder is Unicode. Need to find a workaround

https://github.com/opencv/opencv/issues/8512

moz4r, 2017-11-03:
TODO WORKAROUND : ArrayList to txt map file ?
--> map(Unicode text / clear text )

éकखगàééé > GUID-XXX

moz4r, 2017-11-04:
workaround seem ok, unicode not parsed to screen , but now pushed to publisher
need to polish...

[image]

moz4r, 2017-11-04:
worky, time to sleep again little bug

closed 2017-11-04 by moz4r