MyRobotLab/myrobotlab #85: New sleep tag inside programAb

moz4r, 2017-06-16:
Hi Kevin , do you think it is possible to add a new tag Inside programab.
To put a threaded delay between actions. 

<li><oob>action()</oob></li>
<li><sleep 5/></li>
<li>ok finished</li>

Goal is to convert an globalize some gestures like this one
https://github.com/MyRobotLab/inmoov/blob/develop/InMoov/gestures/lookaroundyou.py

kwatters, 2018-03-03:
i think this would be better hsandled with better sraix support... i'll think about it a bit.  we've had the idea to have a blocking oob call..  and for AIML , i think something like that would be better to be in an sraix tag.

moz4r, 2018-03-06:
yeh good idea, blocking oob will be useful

kwatters, 2018-07-16:
I think the new MrlSraixHandler will address this.  When we create a programab bot we will have control over what it does when it invokes an Sraix tag.  Right now the implementation doesn't do anything other than return a message saying "not implemented" ...  But this should probably be our integration point.

kwatters, 2018-07-16:
I added the SRAIX + OOB support and commented on it here
https://github.com/MyRobotLab/inmoov/issues/117

So, now , if you put an OOB tag inside of the SRAIX tag, it will evaluate the OOB tag and put the result inline in the response.  This will be done in a blocking way.

I think this can be used to accomplish what you are trying to do here.

moz4r, 2018-07-16:
hey nice !!!!! thank you I will test it asap !!! can a classical oob can be blocking also ? ( or a new tag )

kwatters, 2018-07-16:
Yeah. have a look at the comment here:  https://github.com/MyRobotLab/inmoov/issues/117#issuecomment-405381775   it shows the syntax.
Also, I  checked in a unit test for it to validate the syntax of the following aiml.
https://github.com/MyRobotLab/myrobotlab/blob/develop/src/test/resources/ProgramAB/bots/lloyd/aiml/lloyd.aiml#L150

kwatters, 2018-09-28:
Can you explain what happened?  it should have waited for 5 seconds before doing anything. assuming the python exec method is blocking.

What results did you expect? and what actually happened?

kwatters, 2018-10-01:
Python "exec" method by default is non-blocking.
Change "exec" to  "execAndWait"  

https://github.com/MyRobotLab/myrobotlab/blob/develop/src/main/java/org/myrobotlab/service/Python.java#L504

moz4r, 2018-10-01:
Hey, thanks to pointed this out ! 
If I understand <sraix> will add a blocking action for oob, for java land only ?

kwatters, 2018-10-01:
Not sure I completely understand, if you use <sraix><oob><mrl>....</mrl></oob></sraix>   it will wait for the method to be invoked.  

service: python
method: execAndWait
param: code

That is a blocking method on the python service.  Many service in MyRobotLab are non-blocking, so those will just return immediately.

When OOB is embedded inside the SRAIX tag, the framework will invoke the method and block until it returns.  (That method might not block, as is the case with the "exec" method on the python service, that returns immediately and starting running the code on a separate thread.)

moz4r, 2018-10-01:
great, execAndWait and OOB inside the SRAIX tag do the job !
I think I need something else now, I want to get the return of python interpreter. Do you think we can add a new thing like `public String execAndReturnString(String code)` ? 
today exec return a boolean and execAndWait null.

moz4r, 2018-10-05:
Progress ! pAB get the python return statement. I'm stucked in the rabit hole now. 
Sraix blocking the whole template. 
```

<category><pattern>TEST</pattern>
<template>
I want to say something before SRAIX !

 <sraix><oob><mrl><service>python</service><method>evalAndWait</method><param>Sleep(10)</param></mrl></oob></sraix>

</template>
</category>
```