MyRobotLab/myrobotlab #88: BatteryLevel in InMoov3DApp not Happy

supertick, 2017-06-24:
getting index out of range error in START_INMOOV.bat
Is there 101 pictures to display battery level ?  
START_INMOOV.bat seems mostly worky on start .. except for this error and ear webgui is not happy (I'll look into that)

java.lang.ArrayIndexOutOfBoundsException: java.lang.ArrayIndexOutOfBoundsException: 127

        at org.python.core.Py.JavaError(Py.java:546)
        at org.python.core.Py.JavaError(Py.java:537)
        at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:188)
        at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:204)
        at org.python.core.PyObject.__call__(PyObject.java:478)
        at org.python.core.PyObject.__call__(PyObject.java:482)
        at org.python.core.PyMethod.__call__(PyMethod.java:141)
        at org.python.pycode._pyx278.HealthCheck_def$1(C:/mrl.test2/InMoov/life\HealthCheck.py:33)
        at org.python.pycode._pyx278.call_function(C:/mrl.test2/InMoov/life\HealthCheck.py)
        at org.python.core.PyTableCode.call(PyTableCode.java:167)
        at org.python.core.PyBaseCode.call(PyBaseCode.java:138)
        at org.python.core.PyFunction.__call__(PyFunction.java:413)
        at org.python.pycode._pyx280.f$0(<script>:1)
        at org.python.pycode._pyx280.call_function(<script>)
        at org.python.core.PyTableCode.call(PyTableCode.java:167)
        at org.python.core.PyCode.call(PyCode.java:18)
        at org.python.core.Py.runCode(Py.java:1386)
        at org.python.core.Py.exec(Py.java:1430)
        at org.python.util.PythonInterpreter.exec(PythonInterpreter.java:276)
        at org.myrobotlab.service.Python$InputQueueThread.run(Python.java:120)
Caused by: java.lang.ArrayIndexOutOfBoundsException: 127
        at org.myrobotlab.jme3.InMoov3DApp.setBatteryLevel(InMoov3DApp.java:126)
        at org.myrobotlab.service.InMoov.setBatteryLevel(InMoov.java:1991)
        at sun.reflect.GeneratedMethodAccessor94.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:498)
        at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:186)
        ... 17 more

moz4r, 2017-06-26:
Hi ! forced to 100 if overload

supertick, 2017-07-07:
worky ! - Don't see error anymore Thanks moz4r !

closed 2017-07-07 by supertick