MyRobotLab/myrobotlab #89: face recognizer freezes

hairygael, 2017-06-27:
Hello guys.
I think it freezes since we have sub folders in the Training directory.
I'm using this python command, and I tried different options.
https://github.com/MyRobotLab/inmoov/blob/develop/InMoov/gestures/facerecognizer.py

When I comment
```
fr=i01.opencv.addFilter("FaceRecognizer")
fr.train()
```
it doesn't freeze anymore but of course it doesn't recognize the faces anymore.

kwatters, 2017-06-27:
What's in your training folder?  Are you starting with a training set?  The training images will need to be moved into subfolders for each person.

hairygael, 2017-06-28:
Hello Kevin,
Subfolders are created correctly and each subfolder has the name and
contains the various pictures of the person.
Myrobotlab/Training/Gael/pictures
All that works fine.
The freezing is an issue that Dominique has encountered as well.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2017-08-28:
Can this be reproduced ?  What are the steps ?

moz4r, 2017-09-27:
Fresh test of faceRecognizer, it seem Cascade Classifiers wont do the job
Is it worky for you guys ?  Off course I tested other things that the monkey

[image]

`opencv = Runtime.start("opencv","OpenCV")`
`opencv.captureFromImageFile("monkey.jpg")  `
`opencv.capture()  `
`opencv.addFilter("FaceRecognizer")  `
`opencv.setDisplayFilter("FaceRecognizer")  `
`fr= opencv.getFilter("FaceRecognizer")  `
`fr.train()`

hairygael, 2017-10-02:
I tested last night very briefly with version MRL 2556. Face recognizer
recognized me correctly, but I didn't try to add someone new to see if the
clasifier was parsing correctly. My training folder already contains about
8 trained folder of people.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

closed 2017-10-24 by moz4r