MyRobotLab/myrobotlab #91: Cannot stop OpenCV capture

supertick, 2017-07-07:
Hi Moz4r,
I was wondering about this code 

https://github.com/MyRobotLab/myrobotlab/blob/develop/src/org/myrobotlab/service/OpenCV.java#L237-L241

I'm curious to why you added it.  At the moment I'm having problems stopping the capture because of it.  The stopCapture request comes in on one thread, and there is a high likelyhood of resetting it to "true" before the video processing thread checks to see if its done. So, that leads to not being able to stop.

kwatters, 2017-07-07:
Generally the publish methods should only be used for mrl framework stuff ( invoke(...)  )   There really shouldn't be any logic in there , it should just return what's passed to it.

which brings up another question, a few lines down, the publichOpenCVData is marked as void and doesn't return anything...  that also seems kinda wrong to me.

moz4r, 2017-07-08:
Hi Guys ! The idea was to be sure opencv is started and capturing. To set i01.isopencvcapturing=True/False an easy understable var callable even if opencv service is not launched. I can revert and create an other var maybe . Was do you think ?

moz4r, 2017-07-08:
The final goal was to be sure the user camera worky before launching anti opencv related things. About publishOpenCVData marked as void  I didnt change it. you can set it as it must be

supertick, 2017-07-08:
Sure.. No Worries .. I'm just merging, refactoring & cleaning up
For the future:
- capturing is a "control" variable, I changed it from public to namespace protected
- to start and stop capturing the methods capture() & stopCapture() should be used
- isCapturing() returns the boolean value of capturing, which should be accurate regarding state
- if the video thread terminated abnormally,  capturing will be set to false

as for setting it the only methods which should set it are the capture() & stopCapture() methods

I've took out setting the variable from the publishOpenCvData

We're making progress !

moz4r, 2017-07-08:
Thank for your cleanest recommended methods and your work , I will follow.
I like the method to check if there is a worky camera, so we can tell the user there is a problem.
this is the trick https://github.com/MyRobotLab/inmoov/blob/develop/InMoov/services/D_OpenCv.py#L56

supertick, 2017-07-15:
added isCapturing
refactored (too many booleans)
tested capture/stop for OpenCV & VideoInput framegrabbers

closed 2017-07-15 by supertick

hairygael, 2019-04-17:
Hello,
Testing 1.1.159 with inmoov-develop.
With START_INMOOV.bat
Seems openCV starts during initialisation (which is normal) but doesn't stop and this cause a malfunction when trying to launch the capture again.
It also locks MRL when trying to shut down the software.