MyRobotLab/myrobotlab #94: wikidata getdata quantity broken ?

moz4r, 2017-07-12:
what is the fruit of apple tree > ok ( fruit:P1672 )
what is the height of eiffel tower > nok

getData("eiffel tower","P2048");

moz4r, 2017-07-12:
hi beetle, is it worky for you ?

beetlejuiceFR, 2017-07-12:
Hi, No it stopped to work for me too ! I will try to fix it !

moz4r, 2017-07-12:
Thank you ! Happy to "see" you ! I think it crash here : https://github.com/MyRobotLab/myrobotlab/blob/master/src/org/myrobotlab/service/WikiDataFetcher.java#L282

There is a wiki sdk 0.0.8 Snapshot but I didnt test it

beetlejuiceFR, 2017-07-12:
I'm still trying to find the problem . I think more it's caused by the getSnak() function . Strangely it work fine for other queries . At this line https://github.com/MyRobotLab/myrobotlab/blob/master/src/org/myrobotlab/service/WikiDataFetcher.java#L241 , i add the value in an array that contain the dataType ( in A1(0) ) and the value ( in A1(1) ) . For a quantity, A1(1) is empty ! So getSnak found the page "Eiffeil Tower" and the property "P2048" but can't get the value for this property ....

hairygael, 2017-07-12:
> org/myrobotlab/service/WikiDataFetcher.java#L241 , i add the value in an

Just to mention but I'm sure you already know, through ProgramAB, I get the
correct answer regarding the heigth of the Eiffel Tower. But of course it
doesn't use DataFetcher, it searchs the net and get the answer on the
website Answers.com.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

beetlejuiceFR, 2017-07-18:
Hi, sorry but i still didn't find the problem ! Perhaps, the version 0.8.0 of wikidata toolkit will repair this. I'm waiting this new release but i'm still trying to fix it ;)

@hairygael : thx . But you're right both services are different !

supertick, 2017-08-28:
Hi, I'd like to help...  
Can we come up with a set of tests, which we can put into the auto-magical-tests we run every build ?  So that we can see if the service is worky or not worky ?

I'll start searching for example tests - my usual mode of working is to test in the main() function - then move those tests to a Junit class named WikiDataFetcherTest...   I'll start looking for some tests...

moz4r, 2017-09-01:
Thanks , help is welcome !
I have set the main() with noworky sample 
That will break the build :) 

Datatype : quantity01:07:55.090 [main] ERROR class org.myrobotlab.logging.Logging - ------
java.lang.NullPointerException
	at org.wikidata.wdtk.datamodel.helpers.ToString.toString(ToString.java:186)
	at org.wikidata.wdtk.datamodel.json.jackson.datavalues.JacksonValueQuantity.toString(JacksonValueQuantity.java:122)
	at org.myrobotlab.service.WikiDataFetcher.getData(WikiDataFetcher.java:293)
	at org.myrobotlab.service.WikiDataFetcher.main(WikiDataFetcher.java:54)

supertick, 2017-09-14:
Fixed.
I'm impressed - very cool service, I did not find much documentation on google regarding example client.  There were example clients, but what they did are not as clear as this service which beetle created.

So the problem was "units" was being retrieved by this
`String unit = data.toString();`

data is some json object .. the "weird" thing about this is it blows up when you do a .toString() - which usually does not happen in decent code - but somehow this is borked...
data.toString() is dangerous I guess in the wiki-data-client code :(

I did this,
`String unit = String.valueOf(((QuantityValue) data).getUnit());`
Worky
325 **metres** comes back

This is the javadoc with the clue 
http://wikidata.github.io/Wikidata-Toolkit/index.html?org/wikidata/wdtk/datamodel/interfaces/QuantityValue.html

There are other places where data.toString() is used, but I'll guess we'll fix them (when/if) they blow up.

Cheers !

beetlejuiceFR, 2017-09-14:
Well done ! Thank you !

moz4r, 2017-09-14:
nice !! thank you captain

moz4r, 2017-11-22:
seem we need to update Wikidata-Toolkit to 0.0.8
https://github.com/Wikidata/Wikidata-Toolkit/issues/293

moz4r, 2017-11-23:
Fixed & worky locally , pushed but build is broken because Jenkins issue ( need to refresh local git repo )

moz4r, 2017-11-23:
worky !

closed 2017-11-23 by moz4r