MyRobotLab/myrobotlab #96: Missing

moz4r, 2017-07-21:
[image]

Long time I saw @Calamity01 I hope everything is ok for him. 
Look forward the day Inmoov will catch a beer in front or him with next stage, ik computation.
Ses you dude ;)

supertick, 2018-07-29:
I see his name drift through the shoutbox occasionally, hope he's doing well

closed 2018-07-29 by supertick

moz4r, 2018-08-18:
Found him, goodtime !! [image]

Calamity01, 2018-08-20:
LOL... very good time indeed :)

kwatters, 2018-08-21:
Awesome!  Calamity, if you are around in mid October, I will be in Montreal, would love to catch up!

supertick, 2018-08-21:
I'm hoping I'll be there too ! ..
I know I'm in good company when the people around me use github for facebooking ;)

Calamity01, 2018-08-21:
It will be great to meet you :)

hairygael, 2018-08-23:
Great to see you Calamity!
I see Anthony got his MyRobotLab T-shirt and brought you the InMoov hat.
:)

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>