MyRobotLab/myrobotlab #99: No velocity inside virtual arduino

moz4r, 2017-08-12:
velocity seem like always = -1 if using virtual arduino

moz4r, 2017-08-12:
I didnt found how to fix it. someone have an idea ?

supertick, 2017-08-28:
I tested this - I can see when setting Velocity = 3 in SwingGui
[servo] INFO  c.myrobotlab.framework.Service - servoSetVelocity servo id 1 velocity 3
and virtual MrlServo's velocity becomes 3

Perhaps I'm not seeing the issue ?  Can you describe it more moz4r ?

moz4r, 2017-08-28:
Video will explain ! 

Virtual world : https://youtu.be/Xx9lIE1yRGs
Real world : https://youtu.be/363NrwflIWs

hairygael, 2017-08-28:
Great video Anthony,
I was also wondering if it was an issue with only my virtual InMoov or a
general issue.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2017-09-07:
Ah .. I see .. thanks for the video - makes sense now

moz4r, 2017-10-05:
Hi Grog, do you think this can be fixed easily ? This will be very useful Inside test environment

moz4r, 2018-12-18:
waiting cleaner things , I asked i01.left / i01.right to use i2Ccontroller connected on a fake raspi service in virtual mode.
with the help of last servocontroller interface changes inside inmoov . worky !
Not perfect because velocity(-1) in virtual world is very fast, but better than today !

closed 2018-12-18 by moz4r