MyRobotLab/myrobotlab #101: Remove DNA from MyRobotLab - Refactor InMoov services with "attach" function and no DNA

supertick, 2017-08-19:
DNA & Peers was an attempt to "auto-configure" complex services.   But it proved problematic.  Now we are interested in using "attach" functions following our attach patterns

* http://myrobotlab.org/content/attach-pattern
* http://myrobotlab.org/content/service-attaching-diving-deeper
* http://myrobotlab.org/content/more-attach

kwatters, 2017-08-19:
Project Voltron!

supertick, 2017-08-19:
I'll begin by creating an InMoov2 without peers

supertick, 2017-08-27:
will not be doing this for Manticore release - dropping it from project board

moz4r, 2017-10-05:
Is it the way to go ? https://github.com/MyRobotLab/pyrobotlab/blob/develop/service/InMoovEyelids.py

MatsMRL, 2017-10-05:
> Is it the way to go ? https://github.com/MyRobotLab/
> pyrobotlab/blob/develop/service/InMoovEyelids.py

No. I don't think so. You are still specifying that both eyelids are using
the same servoController
I would prefer the attach to be to each servoControl like this
public void attach(ServoControl leftEye, ServoControl rightEye)
That way you would be able to:
1. Have a different ServoController for each Servo
2. Use DiyServo instead of the Servo

MatsMRL, 2017-10-05:
However, the idea of using attach instead of peers is good. So I think that
is the way to go.
As a rule of thumb.
1. Use interfaces.
2. Only attach one level away, not more.
Servo is one level and implements the ServoControl interface. ( Good to
attach to).
Arduino is two levels away and implements the ServoController interface, (
Avoid attaching to this level ).

2017-10-05 23:04 GMT+02:00 Mats Önnerby <mats.onnerby@gmail.com>:

moz4r, 2017-10-10:
Ok, I understand now, there is is ServoControl **+** ServoController 
Is it correct ? ( two methods, 1 open, 1 easy for default things )

1/ `public void attach(ServoController controllerEyeLidLeft, ServoControl eyeLidLeft,Integer eyeLidLeftPin, ServoController controllerEyeLidRight, ServoControl eyeLidRight, Integer eyeLidRightPin)`

->>

`controllerEyeLidLeft = Runtime.start("controllerEyeLidLeft ","Adafruit16CServoDriver")`
`controllerEyeLidRight = Runtime.start("Arduino","Arduino")`
`eyeLidLeft = Runtime.start("eyeLidLeft ","DiyServo")`
`eyeLidRight = Runtime.start("eyeLidRight","Servo")`
`eyeLids.attach(controllerEyeLidLeft, eyeLidLeft, 2, controllerEyeLidRight, eyeLidRight, 4) `

2/ `public void attach(ServoController eyelidsController, eyeLidLeftPin, eyeLidRightPin)`

->>

`eyelidsController= Runtime.start("controllerEyeLidLeft ","Adafruit16CServoDriver")`
`eyeLids.attach(eyelidsController, 2, 4) `

moz4r, 2018-02-27:
New Inmoov V2 in place ! It's a white page of code... 
Let attach things !

moz4r, 2018-06-04:
Drafted InMoovHand2.
We can now use any supported servo & controllers, backward compatibility assured for existing gestures & standardized hardware configurations. If we are ok with this base method, need to extends others.

https://github.com/MyRobotLab/myrobotlab/blob/e39be622638537929a391648afbccef155b0960c/src/main/java/org/myrobotlab/service/InMoovHand2.java#L57

supertick, 2022-07-09:
InMoov2 has been refactored to use pub/sub in combination with new configuration and peer defaults
much has changed

closed 2022-07-09 by supertick