MyRobotLab/myrobotlab #102: OpenCV doesn't work on Ras PI

kwatters, 2017-08-19:
The introduction of opencv 1.3.x has caused OpenCV to fail to capture video on the RasPI.  This update was done to support DL4J which depends on OpenCV 1.3 

The work around on the ras pi , is to download/install everything.  then delete out the opencv specific jars and revert them to opencv 1.1 .. but this is a manual & error prone process.

supertick, 2017-08-30:
arm OpenCV 1.3 still borked?  Does Sarxos frame grabber work ?

CheekyMonkey, 2017-08-30:
The Sarxos framegrabber gives this exception on the Pi 3 under JavaCV1,3

[image]

However, the FFmpeg framegrabber works under JavaCV1.3 on the Pi 3 from Python under mrl:

Python:

frameGrabberType = "org.bytedeco.javacv.FFmpegFrameGrabber";
opencv.captureFromResourceFile("/dev/video0");
opencv.setFrameGrabberType(frameGrabberType);

kwatters, 2017-08-30:
interesting.. looks like sarxos (webcam captur) is using Bridj for some of it's native bindings.  I don't think bridj has the native libs included for the arm pi3 architecture..  hence this error..  there does seem to be a newer version 0.3.11 (webcam capture) and bridj 0.7  .. that might help things..

for now..  FFMpeg seems to be the easiest way to go..  alternatively, using Mjpeg frame grabber, might work pretty well.   (ffmpeg is pretty laggy.. expecially without pyramid down.)

supertick, 2017-09-14:
Worky ?

CheekyMonkey, 2017-09-14:
Hi Grog

This remains an open issue

workarounds: 
* use an alternate framegrabber other than the opencv framegrabber, e.g. ffmpegframegrabber
* revert the libraries for javacv in /libraries/jar/ to javacv version 1.1

CheekyMonkey, 2017-10-19:
Incidentally, is this the same issue being described [here](https://groups.google.com/forum/?nomobile=true#!topic/javacv/tTY6iNHtasU) (with explanation from Samuel)?

kwatters, 2018-04-25:
confirmed fixed in nixie 1.1.103+

closed 2018-04-25 by kwatters