MyRobotLab/myrobotlab #104: Remove all sensor stuff from Servo

kwatters, 2017-08-24:
Servos are not Sensors!!!!

This ticket is to clean up all of the servo stuff.  A finger sensor should be a new service in MRL.. not just wedged into the existing servo service.

Servo services control servos!

moz4r, 2017-08-24:
Ok . Idea was to set a torque to servo by adding a new method.
I Will try to migrate things.
New generic service or inmoov hand mod ?

kwatters, 2017-08-24:
There should be a "Sensor" service that handles this. so a new service.  but we want a code freeze until Manticore is released.  

so. please no new check-ins for the myrobotlab project until after that's done.

supertick, 2017-08-25:
We need Testing ! ..   and since we are in the mode of testing & fixing bugs, we should create unit tests - let the machine test all these things we have fixed in the past ... 

We can use the tickets marked as closed as a reference to create unit tests !

supertick, 2022-07-09:
servos are being auto-tested, sensor code was removed, lots of refactoring

closed 2022-07-09 by supertick