MyRobotLab/myrobotlab #105: Remove All InLine Sensor Code from Tracking

supertick, 2017-08-25:
We need to define a Sensor perhaps similarly to Mats I2C devices.
Being that "Sensors" vary so dramatically as to their dependencies, configuration and output I wonder if a Generalized service would be of much use...

Pin or PinArrays are one of the lowest level sensors & controllers - They currently are not services, and I don't think there is much value into creating a Pin/PinArray service...

However standard interfaces probably should be defined which provide an api to do common things .. like encoding & controlling position

kwatters, 2017-08-25:
I disagree.  I think we should definitely have an analog pin array service.  I mean, why not ?

hairygael, 2017-08-25:
Hello ,
InMoov fingers has a super basic pressure sensor which could be used if MRL
had a sensor service.
[image: Images intégrées 1]
[image: Images intégrées 2]
The sensor is made out of antistatic foam and two wires. The Nervo board
provides the breakout boards on which the user can add a a resistor to pin
up or down at his choice. (Newer version, picture below is the old version)
[image: Images intégrées 3]
The Nervo boards use the analog pins for those sensors.
I would love it they finally could become active and useful.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>