MyRobotLab/myrobotlab #106: Arduino resources {mrl install dir}/resource/Arduino should be extracted "at least" when the first Arduino is started

supertick, 2017-08-27:
.. dunno if it should be done for "every" Arduino instance created - really, I guess it should be done for every new version of the jar - but the jar doesn't know the version of the extraction ... Soooooo, the version.txt should be extracted as part of an extraction, and then compared in the next extraction - if they are different then full extraction forced over all files ..

kwatters, 2017-08-27:
I think this should happen on the first time you start mrl after a clean download.

supertick, 2017-08-27:
Simple & effective logic would be : 
- check if version file has been extracted with all other resources
- if yes - read version - if version == jar version - continue
- if version file not there or version != jar extract everything

kwatters, 2017-08-27:
aye aye captain. sounds good to me.  the real hope is that it's easy for folks to find the mrlcomm.ino ...

supertick, 2017-08-27:
changed commented out Zip.extract to FileIO.extractResources(false) - where it "should" extract on the first time (check's if directory has been created) ...
This is not optimum, but I think it would be an improvement and "good'nuff"  for Manticore release .. I'll remove the ticket from the project board if it tests as expected, but keep the ticket open for "version" improvements

moz4r, 2017-08-27:
Guys some time ago I noticed a difference between linux and windows os. 
Inside linux , dependencies are extracted at every startup ( like inmoov dependencies from repo )
Inside windows, dependencies extracted only at initial installation or update.
What is normal process ?

supertick, 2017-08-27:
Haven't tried it on Linux yet .. but one big change was Runtime's extraction was

1. uncommented (because previously I commented it out - because it was extracting every time on all os's
2. change the extract from Zip.extract - which does not prevent re-extraction, to FileIO.extractResources(false) which does prevent re-extraction after first time...

supertick, 2017-08-28:
I don't think its OS specific moz4r.  I believe you had different versions which behaved differently when you tested.  I just verified it worky on Linux 64 bit same as it does on Windows 64 bit.   It extracts once first time all resources.  If resource directory is found, it won't extract again.  If resource directory is deleted - then it will extract.

closed 2017-08-28 by supertick