MyRobotLab/myrobotlab #108: Service Pages for Services !

supertick, 2017-08-27:
The following services need Service Pages.
Can you help ?

Agent 
AndroidSpeechRecognition 
Arm 
AzureTranslator 
Blocks 
BoofCv 
Chassis 
ChessGameManager 
CsvConnector 
Database 
DatabaseConnector 
DocumentPipeline 
EddieControlBoard 
FileConnector 
GoPro 
HtmlFilter 
HtmlParser 
HttpClient 
ImageDisplay 
ImapEmailConnector 
InMoovEyelids 
Incubator 
InverseKinematics3D 
JMonkeyEngine 
JavaFxGui 
JavaScript 
Lex 
Lidar 
Log 
Mail 
MimicSpeech 
MobilePlatform 
MotorPulse 
NaturalReaderSpeech 
Node 
OculusDIY 
OpenWeatherMap 
Picaxe 
PickToLight 
Pir 
Piston 
Polly 
RSSConnector 
Rekognition 
Ros 
Security 
SegmentDisplay 
SerialRelay 
ServoOrchestrator 
Shoutbox 
Sprinkler 
Sweety 
SwingGui 
TarsosDsp 
TestCatcher 
TestThrower 
TextTransform 
Tutorial 
VirtualDevice 
VoiceRss 
Webcam 
Wii 
WiiBot 
WiiDar 
WikiDataFetcher 
XMLConnector 
YahooFinanceStockQuote 
_TemplateService

moz4r, 2017-08-27:
just a question, how to create a Service Page

supertick, 2017-08-28:
Good question .. you have to be a content admin @ myrobotlab.org
I have anointed you so ...
[image]
You should now have capability of making a service page when you press "create content"
[image]

I tried to standardized the format with forms, but it did not work so well. When you save it you have to go in and edit it again and make sure the "url" is  /service/{serviceName}  vs /content/{serviceName} - and then webgui & swinggui and other things will understand it as "help" for a service..

moz4r, 2017-08-28:
Great ! It is a good solution to test and discover all services !
Lets feed mrl bible !
todo list ( near the top )

supertick, 2017-09-14:
Thanks Moz4r !!! - Lots of bible pages
Your a dedicated monk ;)
https://www.google.com/url?sa=i&rct=j&q=&esrc=s&source=images&cd=&ved=0ahUKEwiqv4Pf56PWAhUC0GMKHaYgAQwQjRwIBw&url=http%3A%2F%2Fwww.victorianpicturelibrary.com%2Fdownloads%2Fmedieval-monk-calligrapher%2F&psig=AFQjCNGp0qeoRf9E4kTx2pm117iiMHYqQw&ust=1505448714856383

supertick, 2017-10-08:
Spectacular mopping from moz4r .. lots of new very nice scripts.  Service page scripts auto-magically added to Service pages.  Good Work ...goodtimes....

closed 2017-10-08 by supertick