MyRobotLab/myrobotlab #109: Enjoyable & Easy Software

supertick, 2017-08-28:
Ease of use is always a high priority for mrl.  So, what things can we do to make the "user" experience more easy or enjoyable ?  Remember the "user" may be yourself ;)

oldfarmhand, 2017-08-28:
Make it so the connections that you need to do something is automatic.
Say you want to connect to Inmoov everything you need opens at one time
like hearing voice sight servos

hairygael, 2017-08-28:
Hello oldfarmhand,
If I understand what you request, it already exist if you use the config
files within MyRobotLab/InMoov/config.
Then just double clik the START_INMOOV.bat file and all is launched at once.
For more info on the setup:
http://inmoov.fr/how-to-start-myrobotlab/

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

oldfarmhand, 2017-08-28:
Thanks Great information!

Kakadu31, 2017-08-29:
Thats a nuce thread GroG, in my opinion the main issue of mrl is the documentation and tutorials which are completely misaing or a few years ols, but with your recent switch one can do from master to develop we should definately create documentation with now more meaning. There shouldaube be an discussion on how we clean up pyrobotlab to get some of these files as tutorials too?

wilcovantoorn, 2017-08-30:
I am trying to get the wiki a bit more up to date, with how to download, how to start and so on.
Because im back after a long time, i really almost have to start from scratch, and there are indeed few tutorials and documentation.
So im writing down what i can find on myrobotlab and on the inmoov site, and put it on the wiki.
That is, if you want me to.

moz4r, 2017-09-06:
[image]

Inmoov related things, relayed from French forum...

- "We need Enjoyable & Easy thing to create gestures"
> I think this is gesture Creator !!
Ok after manticore we can refresh and add some things like roll neck, neopixel, voices widget ...

- X/Y coordinates from face tracking to follow recognized person ( not related to the subject but noted )
- openweathermap forecast enhances ( not related to the subject but noted )

Personally : 
- I think we can do something with Scratch service ( not necessarily for Inmoov ), I d like my kids use MRL too !!

- An assistant at first MRL start

- A GUI FOR INMOOV !
subject and discussions already opened,
But waiting the dday maybe we can polish/refactor existing things ( Inmoov SwingGui service with the help of some json files related to some services ) .

- Update MRL and take care of related dependencies like mrlcomm !
Already Implemented some things with the help of batch launcher > TODO use the agent

and of course what is Inside the 1st post

Dom1416, 2017-09-10:
Always the last line in the python tab.

Dom1416, 2017-09-10:
Coordonate XY in the face recognizer data to make tracking in the same time of face recognizer

Dom1416, 2017-09-10:
restart MRL with some command like "Runtime.reboot" or "Runtime.restart"

supertick, 2017-09-10:
@moz4r 

> I think we can do something with Scratch service ( not necessarily for Inmoov ), I d like my kids use MRL too !!

FYI - http://myrobotlab.org/content/blocks-mrl

supertick, 2017-09-10:
Runtime.restart should work next release

The Python editor really needs to be a better editor in regards of saving and maintaining file references, menus & examples

hairygael, 2017-09-11:
You shouldn't ask me...
:)
-Cloud sharing for InMoov related to face tracking and face recognition
with AIML personality subfolder associated. (We need to explicitely set and
think of a permission and moderation organism, I have some lawyers that are
interested to participate)
-Kinematic for object grabing(Kevin started on that)
-Object recognition via machine learning(Kevin made a great sample video)
-Extra PID for the rollneck of InMoov(PID1-neck, rothead, PID-2-eyeX,eyeY,
PID3-rollneck)
-Lidar compatible (I will get one soon for to test, the company is
providing me one)
-Option to choose other devices than Kinect from Microsoft for skeleton
tracking via OpenNI. (I have an Orbbec Astra which doesn't seem to react
when tested with latest MRL)
-Fix Gesture Creator and enhance for easier use. (Currently a bit complexe
and missing documentation for Newbie)
-Start to think of a full PID control center for the legs, how do we
proceed? Shall the computing be done on a seperate arduino for each leg and
send back to MRL for final computing?
-Segway implementation for robot. Autobalance mobile platform is something
I seriously want to built for InMoov for fast driving, but it also needs to
be stable if not in autobalance mode, that is why, I want to create a
Segway with 6 wheels (3 on each side triangulated), which could auto lock
in a stable mode position and in my dreams, climb stairs.

Oh man I could keep going with the list...
If I think of more simpler stuff, I will add them.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2017-10-08:
I want to be warned via a flashed message in either gui that something went wrong - and then someplace i can easily click and have more details 

I started work in the webgui & swinggui to present "warn/info/error" messages - but there needs to be counters of each - angular logging in chrome works this way - so you can easily see if an error has occurred even if the latest detail is about "info" level logging

supertick, 2017-10-30:
All great ideas ! ... 
Most are "enhancements" which is great ! .. and I'm excited to start implementing them.
Manticore needs only testing & fixes, so I'll keep this ticket but remove it from the Maticore release

Kakadu31, 2017-11-04:
This is my “Vision” for the postmanticore Myrobotlab. It may contain some ideas that are already mentioned previously.

**The Problems I see:**

I think we all agree a big problem at the moment is documentation and the ease of use for some more complex services. The documentation is getting way better at the moment trough the service pages + example scripts. Still for the more complicated services I think its just too complex. In the shoutbox one reads very often the same questions or new thinks nobody seems to really know about, but the veterans.

**The solution?**

The first goal to achieve would be **consistency**. When I want to attach a service I don’t want to think about HOW to connect them. The way should always be: service.attach(“otherservice”) this should in the best case always be an easy pipeline. When I use rogram AB I only want to give an Input and get an Output, same for a lot of Services. This Input and output format should be when possible always the same. In the current Version this would be something like onOpenCVData(“Data”).  This “Data” should contain when possible all the information I want to get: for example the x,y coordinates for a recognized face and its name. In the case of program AB it should be the string of the Answer. If there are some reasons to have multiple output functions they should all be regrouped somewhere and clearly visible. 
Personally I really like the new Service Matrix, maybe one could add a column Input and an Column Output to see all the possibilities? Or even better for each service there should be a list of all core functions one can use in Python. Similar like it is in the runtime tab at the moment. The Problem with the runtime tab is: It is just confusing, there are sometimes too much functions and How should I as newbie know what they do? Most example scripts only use 1,2,3 examples but despite them a Noob has no chance to see what the service can do. Often these examples have a complex structure because of the current publish-listener structure. 
So afterall we want (or atleast I do ;) ):

- Easy Input Output structure with: servie1.attach(“service2”) 
 	-> service1 gets the output of service 2

- A place with all the possible output methods(and even better the other methods, mostly inputs) one can use. This should never be more complicated that a simple method. 

- Ideally the data should be passes in a convenient format like a simple string or array of strings etc. So one could just do something like: OnOpenCvData(“data”) with print data and just read the output so see where the component one want to use hides. Or even OnRecognizedFace with a small specific Output 

- Ideally the program can be reconstructed with a minimum of wires betwees pins on a pinboard.

- If a service has too many methods which are not very usefull from python end, they should not confuse new users and be more hidden

_Ideally we want something as simple as this (only very roug example not actual code):_

Create arduino
Create motor1 and attach to arduino
Create opencv
Create speaker
Create ear
Create ProgrammAb

Ear use device zzzz
Opencv use grabber xxx and start
Motor1 use pin 6,4 from the arduino

def onOpenCVData(data):
	if face = recognized
		print face.x + face.y
		motor.dontmove
		speaker.say(“hello”)
	else 
motormove

def onEarData(data):
answer = ProgrammAb.getAnswer(“data”)
	speaker.say(answer)

supertick, 2017-11-06:
Great comment Kakadu.
I think we as a collective have come a long way in beginning to standardize the "attach" methods.
Attach is two way not one.  Often both services need to do some necessary activity before they are both attached.  If they are coded in the correct way, these activities are accommodated, additionally the users are allowed not to have to remember service1.attach(service2) or service2.attach(service1)  .. they both work.

I like where you mention you'd like to see all the "output" of a service.  This could pretty easily be done by reflecting all "publish" methods.  A very good idea - just not sure where to display it.

Kakadu31, 2017-11-06:
We could Display it either in the Service Pages, or do a new table for it similar to the Service Matrix. It would be really nice to have them in one place so as beginner one could check this place and see:

onOpenCVData
onFaceRecognized
onHeardSomething
onJoystickData
etc..

So there would happen 2 things:
- First one sees immediately what "sensorData" you can immeditely use in a myrobotlab programm
- Second one would immediately see how to do it as it shouldn't be much more complex than creating the service, call the method, and treat the data the right way. For every "Output" we could write a 4-5 line example which does something like getting the x and y coordinate of a box around a detected face or get the name of a recognised face etc..