MyRobotLab/myrobotlab #113: Rssconnector / Solr broken

moz4r, 2017-09-04:
I can't launch rssconnector to test / document . It seem Solr is broken
Added httpmime-4.5.2 into solr and httpclient but no effect, any idea ? 

Exception in thread "main" java.lang.NoClassDefFoundError: org/apache/http/entity/mime/content/ContentBody
	at org.myrobotlab.service.Solr.startService(Solr.java:249)
	at org.myrobotlab.service.Runtime.createAndStart(Runtime.java:295)
	at org.myrobotlab.service.Runtime.start(Runtime.java:1730)
	at org.myrobotlab.service.RSSConnector.main(RSSConnector.java:96)
Caused by: java.lang.ClassNotFoundException: org.apache.http.entity.mime.content.ContentBody
	at java.net.URLClassLoader.findClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	... 4 more

So I tried to add

supertick, 2017-09-04:
I suspect clashing HttpClient versions
(something probably we could take care of more easily in the future maven build)

kwatters, 2017-09-04:
We will fix it after the release.  Also update the version of solr.

kwatters, 2018-04-11:
changing out the "feed4j" library with "rome"

kwatters, 2018-04-26:
tested in nixie 1.1.104 and it looks good now. closing issue.

closed 2018-04-26 by kwatters

aroopganguly, 2018-06-08:
any solutions to this ?

LunDev, 2018-06-08:
@aroopganguly 
According to @kwatters this was fixed in April already.
You may want to open a new issue and describe your exact problem there.

kwatters, 2018-06-08:
This is fixed on the Nixie release when we updated the Ras connector to use the Rome  library.