MyRobotLab/myrobotlab #114: Start - stop service

Dom1416, 2017-09-10:
I try this test but doesn't work. Same problem with opencv.

from org.myrobotlab.service import Runtime

openni = Runtime.createAndStart("Kinect", "OpenNi")
print "service kinect ON"
sleep(20)

openni.releaseService()
print "service kinect OFF"
sleep(20)

openni = Runtime.createAndStart("Kinect", "OpenNi")
print "service kinect ON"
sleep(20)

openni.releaseService()
print "service kinect OFF"
sleep(20)

openni = Runtime.createAndStart("Kinect", "OpenNi")
print "service kinect ON"

The first release work but the start again doesn't work. May be it is the same for all services.

Dom1416, 2017-09-19:
No body work for this bug ?

moz4r, 2017-09-21:
I will try this as soon I replug my Kinect. what is the goal ? stopcapture() is not enough ?

Dom1416, 2017-09-21:
Salut Anthony,

Non la fonction stopcapture n'est pas suffisant. En faite c'est la fonction releaseService qui ne va pas. Je pense que le bug apparaît pour tous les services.

Ici le but est de pouvoir couper complètement la kinect lorsqu'elle n'est pas utilisé afin d'économiser l'énergie. Pour un robot de démonstration ce n'est pas grave. Pour un robot qui fonctionne 24/24, certains services doivent être complètement retirés puis redémarré lorsqu'on en a besoin.

Merci de ton aide.

Dom.