MyRobotLab/myrobotlab #118: Fix Mr Turing - make him better !

supertick, 2017-09-15:
Mr. Turing's capability of getting information from the internet is currently borked ...  this prevents srai requests from resolving, and pictures and what not.

But, to make Mr. Turing better he should be able to reply regarding "Help me" questions from the site.
- "How do you start myrobotlab ?"
- "Where can I download myrobotlab ?"
- "My servos aren't working"
Relevant links and explanations would be nice.

Additionally kyle mentioned he should be able to return current build, list of fixes, recent changes and other build information.

Cyber-One, 2017-09-15:
Maybe say hello to registered users when they enter the shout-box!

supertick, 2017-09-15:
or conversation starters like ..
"Hello Ray, did you hear about Cassini deliberately being driven into Saturn by Nasa ?"

Cyber-One, 2017-09-15:
Exactly

supertick, 2017-09-15:
Shoutbox needs pagination too - it should only load one "page" but you should be free to step back and forward all you want.  Mr. turing should also index all the shouts

supertick, 2018-04-08:
I think kwatters wanted to do this one ...