MyRobotLab/myrobotlab #120: Adafruit16CServoDriver servo control 'glitch'

CheekyMonkey, 2017-09-15:
There appears to be an initial "extreme" server move after connecting a servo to an Adafruit16CServoDriver in mrl whch only occurs the first time that the servo is commanded to move

Drilling down with the Eclipse debugger, it appears that the initial extreme server move occurs somewhere in the run() method in this class:

org/myrobotlab/service/Adafruit16CServoDriver:

in
public class SpeedControl extends Thread

in run()

when the method is called from within the servoMoveTo() method at this line
servoData.speedcontrol.start();

Due to the precise "millisecond" timing used in the method, it's challenging to isolate the root cause during a debug session

I'm able to consistently reproduce the condition with the following Python script in mrl

adaFruit16c = Runtime.createAndStart("AdaFruit16C","Adafruit16CServoDriver");

#start a Raspberry Pi instance
raspi = Runtime.createAndStart("RasPi","RasPi");

#attach the AdaFruit16C I2C servo driver to the Raspberry Pi
adaFruit16c.setController("RasPi","1","0x42");

#set the frequency for the AdaFruit16C I2C servo driver to 50 Hz
adaFruit16c.setPWMFreq(0,50);

servoPin = 0
servo = Runtime.createAndStart("servo","Servo");
servo.attach(adaFruit16c,servoPin,70,20);

Run the script above and then use the servo GUI to move the servo.
A "click" anywhere on the servo position slider is sufficient to reproduce the issue

Note the initial servo "extreme" movement, before the servo "settles down"

MatsMRL, 2017-09-17:
The problem is that there is no way for the servocontroller to know the servos "current" position when the move starts. So is assumes that the servo is at the position it should be at after the first moveTo().
So I think the problem is that the servo service returns 0 instead of the rest pos

MatsMRL, 2017-09-17:
Doh. I didn't intend to close. Just FYI. Making a new test.

MatsMRL, 2017-09-17:
I think I found the problem. Please test. Build 2512 or later.

closed 2017-09-30 by MatsMRL