MyRobotLab/myrobotlab #123: Servo methods pushed to ServoControl

supertick, 2017-09-21:
There needs to be some refactoring/standardization of the Servo methods.
We now have awesome autoEnable/Disable functionality YAY !

But the following should be rectified (2 removed & 2 combined) into possibly a single method call with boolean parameter

* enableAutoEnable(boolean autoEnable)
* enableAutoAttach(boolean autoAttach)
* enableAutoDisable(boolean autoDisable)
* enableAutoDetach(boolean autoDisable)

my preference would be to remove the previous 4 and replace with the following one

setAutoEnable(boolean autoEnable)

this method should also get promoted to the interface I think ?

moz4r, 2017-09-21:
Agree

setAutoDisable(boolean autoDisable)
setAutoEnable(boolean autoEnable)

If we want only 1 single method, I would have said 

setAutoDisable(boolean autoDisable)

Because I think there is no logical reason to play with autoEnable and set it to false.
So we can set it to true by default and remove setAutoEnable method.

autoDisable is different. We need to tweak it

supertick, 2017-09-21:
Thanks moz4r..
can you review (do not accept) - my pull request ?

also - what is this for ?

    if (lastPos == pos) {
      if (autoDisable) {
        delayDisable(defaultDisableDelayNoVelocity);
      }
    }

supertick, 2017-09-21:
In the shoutbox Mats kwatters & myself discussed at length ...
Mats agreed with condensing the method to setAutoDisable(boolean autoDisable) because noobie users are more likely to think, "a servo should move to where I tell it ... but later think - hey setAutoDisable .. I wonder if that will save my servos..."  

kwatters at this time has not replied to the shoutbox regarding the name

kwatters also offered to do a pull request - to give an example of reducing the "backend" variables which support the method

I'd like input from moz4r, Mats & kwatters regarding weather to deal with this refactoring this release.

Mats was concerned that InMoov scripts would need to change ...  I don't think so - I'm suggesting we add the "final" method & implementation, but leave all the existing methods as Deprecated.

I would prefer we do this change in this release - my motivation is to show a really good example of how to use the Servo in the /service/Servo.py script.  In order to do this, it should be implemented.

I do not want to do the implementation as I am least experienced in this particular functionality, and any real world experience.  I'll leave it to moz4r, Mats and kwatters to decide on how & who will implement.

Thanks.

kwatters, 2017-09-21:
I think it's clear to explain to people that   setAutoDisable(True) saves servos. 
Let's depricate the autoEnable stuff.  any only have AutoDisable..  where implicitly it will enable the servo when you ask it to move.

moz4r, 2017-09-21:
We re all agree, let's depricate ! 

Grog about your comment :

If servo reach the final position, we can disable it. 

Two cases 
- We know velocity of servo, no problem, we can disable it
- Velocity = -1 ( in this case we dont know when the servo reach position ) : we apply a generic timer to disable ( 10 second by default )

kwatters, 2017-09-21:
If velocity is set, we should move to desired position (we know when we are expected to arrive based on velocity.. ) and then we should hold that position for at least a few extra seconds. 

Otherwise, the InMoov will put people in a head lock when you say "picture on the left side"  ...

I like a generic 10 second disconnect.  so long as it's configurable.

supertick, 2017-09-21:
Awesome ! 
I'm glad we all agree - it be nice if Mats verified by a comment on the ticket.

So next question is who is going to do the commit, so we can wax and polish the remaining /service/Servo.py script of how to correctly use this feature ?

To be clear too - all other methods relating to this will be "marked as deprecated" - so that at least this release InMoov scripts will not break.  correct ?

moz4r, 2017-09-21:
About "extra" enable we have two things

1/
https://github.com/MyRobotLab/myrobotlab/blob/develop/src/org/myrobotlab/service/Servo.java#L1201
You right we need to expose disableDelayIfVelocity to tweak

2/ StartedGesture / StopGesture
https://github.com/MyRobotLab/myrobotlab/blob/develop/src/org/myrobotlab/service/InMoov.java#L1763

Goal is to disable "autodisable" until gesture is finished. about this point some codes line need simplicity ( temporaryStopAutoDisable )

MatsMRL, 2017-09-21:
I agree to setAutoDisable(boolean) and that servos should always be auto enabled at moveTo.
All agreed methods should be in the ServoControl interface, so that they can be implemented also in DiyServo.
The goal is that you should be able to replace any Servo service, with the DiyServo service.
That will require that the InMoov services use the ServoControl methods and not the Servo service methods. But I think we should get Manticore out before that major rework.

Cyber-One, 2017-09-21:
I know I' late to this party, but what about naming it setAutoPower(true/false) When set to auto it turns on and off as required, and does not hit at being one way only...
It's just a thought.
As an overlay function (remeber I'm not a java programer) setAutoPower(false, true) where the second parameter could specify if the servo should be enabled after the auto function is turned off.

supertick, 2017-09-22:
Someone with an InMoov must commit these changes .. I am unworthy .....

moz4r, 2017-09-22:
done, I need testing ( and testing ) before commit

moz4r, 2017-09-22:
"There is insufficient memory for the Java Runtime Environment to continue" . we will wait a little for jenkins rest()

supertick, 2017-09-23:
add more memory to Jenkins ...  should be good 

Looks much cleaner .. thanks !

moz4r, 2017-09-23:
hey thanks for upgrade !!!

supertick, 2017-09-23:
Now we need multiple-InMoovers to test !

kwatters, 2020-05-23:
I think there is a lot of activity on this now in last_servo_fixes branch.  I suspect that merge can/should close this issue.

supertick, 2021-12-20:
long fixed ago ....

closed 2021-12-20 by supertick