MyRobotLab/myrobotlab #124: Servo - temporaryStopAutoDisable ?

supertick, 2017-09-21:
Is this needed ?
What was the intention ?
Why is it temporary ?

moz4r, 2017-09-21:
Hi, yeh it is useful, but it must be moppep for simplicity.

Explainations :

- User set servo.setAutoDisable to true : servo is always autodisable now.
Sometime we dont want servo auto disable ( tracking, some gesturess ... ) So we temporary stop AutoDisable. But after action we want setAutoDisable  go back to original state, set by user.

Exemple : 

https://github.com/MyRobotLab/myrobotlab/blob/develop/src/org/myrobotlab/service/InMoov.java#L1771
https://github.com/MyRobotLab/myrobotlab/blob/develop/src/org/myrobotlab/service/InMoov.java#L1782

moz4r, 2017-10-02:
mopped https://github.com/MyRobotLab/myrobotlab/blob/develop/src/org/myrobotlab/service/interfaces/ServoControl.java#L299

closed 2017-10-02 by moz4r