MyRobotLab/myrobotlab #131: DIY Servo service

CheekyMonkey, 2017-10-01:
In the DIY servo service it appears that any change to the servo max/min limits either through Python or through the DIYServo GUI is not actually getting set

As at development build 2555, looking at the /myrobotylab/src/org/myrobotlab/service/DIYServo.java source it appears that the DIYServo setMinMax() method does the following:

#1
	public void setMinMax(double min, double max) {
	        mapper.setMin(min);
		mapper.setMax(max);
		broadcastState();
		
	}

but maybe ought to do the following:

#2
	public void setMinMax(double min, double max) {
	    map(min, max, min, max);
	}

A preliminary check shows that approach #2 seems to be worky, but looking for your thoughts and guidance on this

MatsMRL, 2017-10-01:
When I made DiyServo I started with copying the Servo service. The Servo service has changed behaviour since that.
I changed the setMinMax method to behave the same way as it does in Servo.
It should work in version 2558.

CheekyMonkey, 2017-10-01:
@MatsMRL , Nice one, that change to DIYServo.java has made a difference (build 2558)

I now see servo min/max reflected in the GUI when I issue a servo.setMinMax(a,b) command from Python, but I also see what appears to be additional motor move commands in the logging once setMinMax() executes

eg
23:05:17.882 [diyservo.motorUpdater] INFO  c.myrobotlab.framework.Service - diyservo.motor info diyservo.motor.move(0.565)
23:05:17.939 [diyservo.motorUpdater] INFO  c.myrobotlab.framework.Service - diyservo.motor info diyservo.motor.move(0.566)
23:05:17.995 [diyservo.motorUpdater] INFO  c.myrobotlab.framework.Service - diyservo.motor info diyservo.motor.move(0.566)
23:05:18.050 [diyservo.motorUpdater] INFO  c.myrobotlab.framework.Service - diyservo.motor info diyservo.motor.move(0.566)

It seems to be a continuous stream of motor move commands
I haven't plugged power into he motor yet...just in case :)

MatsMRL, 2017-10-06:
Commented out a lot of info messages from the AbstractMotor service.

closed 2017-10-06 by MatsMRL