MyRobotLab/myrobotlab #136: I2C servo wont enable after disabled

moz4r, 2017-10-22:
tested on raspi

https://github.com/MyRobotLab/pyrobotlab/blob/develop/service/Adafruit16CServoDriverPi.py

moz4r, 2017-10-23:
I got my hands on the hive, need to deal with other things related..
> remember servo position at restart
> add servo event to i2C servo driver
> set double for every servo things ( sometime conditions failed because 170 != 170.0001 ... )
> fix fake messages sent by servo events ( servo is really moving > isMoving==False )
etc ...

closed 2017-10-31 by moz4r

supertick, 2017-11-01:
Heh.. "hands on hive" ... Every day my RESPECT depends for you moz4r !

juergmaier, 2017-11-01:
hands on hive?? could you elaborate on this idiom?  and your respect would rather deepen then depend?
sorry, not blaming just not beeing able to follow.

moz4r, 2017-11-01:
Hi mates ! What I meant is I wanted to fix one thing, I was thinking of something simple, but I encountered some obstacles... 
Now I think everything fixed , yehhh : 

- Now , when you close mrl servo position is "recorded" inside json. When you open it again. So no more fullspeed movement. Not related to I2C, general thing.
- I2C is worky now like Arduino controller about servo events ( Autodisable, isMoving, Blocking moveto ... )

[image]

supertick, 2017-11-02:
@juerg .. yes, deepen. Inaccurately, autocorrected ... I blame the robot in my phone.